prusa3d/PrusaSlicer #6288: PrusaSlicer Crashes when Slicing Specific Model/Scaling Combination

marius137, 2021-03-29:
### Version
Version 2.3.0+linux-x64 (AppImage)
### Operating system type + version
I am running Archlinux
```Linux archbox 5.11.10-arch1-1 #1 SMP PREEMPT Fri, 26 Mar 2021 00:11:29 +0000 x86_64 GNU/Linux```

### 3D printer brand / version + firmware version (if known)
Anet A8 Plus
Bigtreetech SKR 1.4
Marlin Bugfix

### Behavior
When trying to slice this specific Model and Scaling Combination PrusaSlicer Crashes.
Slicing Works when the Model is scaled to 100%
Slicing also works in Simplify3D when scaled to 200%

#### Project File (.3MF) where problem [occurs]
https://filetransfer.io/data-package/zoDJEj8z#link

combolek, 2021-03-30:
FWIW, I cannot reproduce it with PrusaSlicer-2.3.0+linux-x64-GTK3-202101111352.AppImage
[image]

marius137, 2021-03-30:
I also played around a bit and it works on my Windows 10 VM. Maybe some weird issue with ArchLinux?

rtyr, 2021-03-31:
I am unable to reproduce the issue on win64.

willyb90, 2021-04-01:
I have the same issue on Archlinux. Certain models will crash seconds after slicing.

marius137, 2021-04-04:
is there some way to start prusaslicer in verbose mode or enable logging?

cake808, 2021-04-05:
@marius137 
Just run `prusa-slicer --loglevel N` 
N can be 0 (fatal), 1 (error), 2 (warning), 3 (info), 4 (debug) or 5 (trace)

marius137, 2021-04-05:
I tried launching it with loglevel 5. this is the resulting log https://filetransfer.io/data-package/3v011HKU
at the end it just crashes with 
```
[1]    4451 segmentation fault (core dumped)  prusa-slicer --loglevel 5 | 
       4452 done                              tee prusa.log
```

combolek, 2021-04-05:
Are you using GTK2 or GTK3? Also, can you try PrusaSlicer-2.3.0+linux-x64-GTK3-202101111352.AppImage ?

marius137, 2021-04-06:
the problem exists in the AUR Version, `PrusaSlicer-2.3.0+linux-x64-GTK3-202101111352.AppImage` and in `PrusaSlicer-2.3.0+linux-x64-202101111322.AppImage`. for GTK I have both `gtk2 2.24.33-2` and `gtk3 1:3.24.28-1` installed

combolek, 2021-04-06:
Hmm, so this probably really is ArchLinux specific. It could be the same issue as #6313

marius137, 2021-04-06:
Yep, from what I can Tell it is the same issue. Probably some Update in archlinux Broke it because it used to Work flawless until i updated my system a few hours before I opened the issue.

marius137, 2021-04-12:
the fix mentioned in #6313 (up/downgrading mesa) worked.

closed 2021-04-12 by marius137