prusa3d/PrusaSlicer #6290: Slicer creates holes in sliced model

sadovsf, 2021-03-30:
### Version
2.3

### Operating system type + version
macOS Big Sur 11.2.3

### 3D printer brand / version + firmware version (if known)
MK3S+

### Behavior
When slicing model i downloaded from internet slicer created vertical empty lines on brick wall
Here is a model view:
[image]

Here is what slicer does even with enabled detection of thin walls:
[image]

#### STL File where problem occurs
[Tardis.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/6226628/Tardis.stl.zip)

rtyr, 2021-03-30:
The model needs to be repaired. It will be sliced correctly then.

sadovsf, 2021-03-30:
May i ask what type of fixes it needs? Rendering seems OK, I have step file as well so i may try. Are there some details to debug whats wrong?

rtyr, 2021-03-30:
This video should help you https://www.youtube.com/watch?v=YfkSE-pKSUw

rtyr, 2021-03-30:
Repaired model:
[Tardis_repaired.zip](https://github.com/prusa3d/PrusaSlicer/files/6227752/Tardis_repaired.zip)

sadovsf, 2021-03-30:
Thank you so much, iam still waiting on Windows to process file :D

closed 2021-03-30 by sadovsf