prusa3d/PrusaSlicer #6293: Why the raft feature is so bad?

VijitCoder, 2021-03-30:
### Version
Version 2.3.0+linux-x64

### Operating system type + version
KUbuntu 19.10

### 3D printer brand / version + firmware version (if known)
It doesn't matter. The problem in the slicer itself.

### Behavior
I tried to understand how to use the raft in this slicer. I familiar with Cura and IdeaMaker so I have some opinion about how this feature can be. But.. not in this slicer. 

First of all, why you bound rafts with supports? There is no explicit relation, these features can be used independently in the real printing. But, in the PrusaSlicer, if I need a raft I must to switch "Generate support material = on". Why?!

Ok, let it be so.

Now take a look at the settings and to result of the raft generation:

[image]

[image]

`Interface layer  = 1` - ignored.
`pattern = honeycomb` - applies only to the lower half of the raft.
`Interface loops = true|false` - doesn't matter, nothing changes.
`raft margin` - not configureable and applies to the first layer only.
First layer - not configureable something. The only thing that I found for it - `Extrusion width -> first layer` which in the normal logic shouldn't effect to the _raft fist layer_.

What the heck is this? Why it so terrible? Just take a look to other slicers. There the rafts have clear predictable settings. 

In fairness, I should note that in general, this slicer is noticeably better than the others.

#### Project File (.3MF) where problem occurs
[cube 20 h7.zip](https://github.com/prusa3d/PrusaSlicer/files/6230429/cube.20.h7.zip)

rtyr, 2021-03-30:
Raft will be improved in the next PrusaSlicer release. See https://github.com/prusa3d/PrusaSlicer/issues/3425. Closing.

closed 2021-03-30 by rtyr

VijitCoder, 2021-03-30:
Yeah, now I see.. 

I also found another issue - [# 2767](https://github.com/prusa3d/PrusaSlicer/issues/2767) _"Please Implement High Quality Raft"_, dated Aug 15, 2019. It was 1.5 year ago. You guys really fast developing this slicer. Sweet.

VijitCoder, 2021-03-30:
Oh.. As I see, you make one minor release per year. At least between 2.3.0 and 2.2.0. That's explaining a lot.