prusa3d/PrusaSlicer #6295: PrusaSlicer filing monothread

SckyzO, 2021-03-30:
### Version
2.3.0

### Operating system type + version
windows 10 version 20H2

### 3D printer brand / version + firmware version (if known)
Artillery X1

### Behavior
I change my CPU to a new old processor : E5 2698 V4 (20 Cores, 40 Threads). But when i slice any model (screen 1 and 2), when filling action come, PrusaSlicer use only 1 core/ 2 threads (screen 3). And for any slice, when my model is big, prusaslicer generate my model with lot of time. IdeaMaker/Cura/S3D generate my model in 10 secondes ... but prusaslicer generate in ... 5/10 ... minutes ... 

#### Project File (.3MF) where problem occurs
Any files

=============
if it is a known bug or if I have a problem?

I can dump log or share more screenshots if you want. 

screen 1 
my current model
[image]

Screen 2 
step 1 before filling step : multicore/threads
[image]

screen 3
cpu process in filling 
[image]

rtyr, 2021-03-31:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2263. G-Code export parallelization is on our todo list.

Closing.

closed 2021-03-31 by rtyr

SckyzO, 2021-03-31:
Hello @rtyr 

Thx for your answer. 
Do you think it will be available for next release ? 

Regards,

AnthonyBerduck, 2023-07-09:
Hello @rtyr !

Any updates about multithreading G-Code Export ? It's been a while and we're all waiting for much faster slicing.

Regards,

rtyr, 2023-07-09:
https://github.com/prusa3d/PrusaSlicer/issues/2263#issuecomment-927088470