prusa3d/PrusaSlicer #6303: 'Object Outline' is under BRIM'

cjshaker, 2021-04-01:
Printing a dozen filament clips from "Yet another filament spool clip" by streefland on Thingiverse: https://www.thingiverse.com/thing:4694015

The object outline gets drawn UNDER the Brim!
[image]
[PrusaSlicer.zip](https://github.com/prusa3d/PrusaSlicer/files/6245845/PrusaSlicer.zip)

### Version
Version 2.3.0+win64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Edition	Windows 10 Home
Version	20H2
Installed on	‎6/‎17/‎2020
OS build	19042.906
Experience	Windows Feature Experience Pack 120.2212.551.0

Device name	CJSHAKER
Processor	Intel(R) Core(TM) i7-3820 CPU @ 3.60GHz   3.60 GHz
Installed RAM	16.0 GB
Device ID	FFE43E10-A1F5-4992-8FC7-814CF3D83E0D
Product ID	00326-10000-00000-AA616
System type	64-bit operating system, x64-based processor
Pen and touch	Touch support with 10 touch points

### 3D printer brand / version + firmware version (if known)
Prusa Original i3 MK3S+, firmware version 3.9.3
[image]

### Behavior
I imported the STL file. Used Prusa Slicer to make 12 copies of the filament clip.
Enabled Brim printing.
I expected the object outline to be outside of the Brim, but it got drawn UNDER the brim!
I included a photo, and two screen shots of the problem. You can see it on the Prusa Slicer preview.
filament-clip.zip is a Zip file of the project file.
[filament-clip.zip](https://github.com/prusa3d/PrusaSlicer/files/6245874/filament-clip.zip)

[image]

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2021-04-01:
Intended behavior. We may change it in the future though as it is quite frequent topic.

Why this behavior was implemented:
#724

Duplicate issue + links to other similar issues:
#3692

Side note. Looking at your photo, you Live Z is not set correctly (the nozzle is too high).

Closing.

closed 2021-04-01 by rtyr

cjshaker, 2021-04-12:
If you have a Brim configured, why do you also need an Object Outline? Doesn't a Brim fill that function as well?