prusa3d/PrusaSlicer #6304: Slicing Preview disapear

hisashime, 2021-04-02:
2.3.0+

PrusaSlicer
Version:   2.3.0+
Build:     PrusaSlicer-2.3.0+-202101111304

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS High Sierra Version 10.13.6 (Build 17G14042)
Total RAM size [MB]: 4,295MB
OpenGL installation
GL version:   2.1 INTEL-10.4.14
Vendor:       Intel Inc.
Renderer:     Intel HD Graphics 3000 OpenGL Engine
GLSL version: 1.20
Installed extensions:

GL_APPLE_aux_depth_stencil
GL_APPLE_client_storage
GL_APPLE_element_array
GL_APPLE_fence
GL_APPLE_float_pixels
GL_APPLE_flush_buffer_range
GL_APPLE_flush_render
GL_APPLE_object_purgeable
GL_APPLE_packed_pixels
GL_APPLE_pixel_buffer
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_specular_vector
GL_APPLE_texture_range
GL_APPLE_transform_hint
GL_APPLE_vertex_array_object
GL_APPLE_vertex_array_range
GL_APPLE_vertex_point_size
GL_APPLE_vertex_program_evaluators
GL_APPLE_ycbcr_422
GL_ARB_color_buffer_float
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_instanced_arrays
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_provoking_vertex
GL_ARB_seamless_cube_map
GL_ARB_shader_objects
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shadow
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_non_power_of_two
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_transpose_matrix
GL_ARB_vertex_array_bgra
GL_ARB_vertex_blend
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_window_pos
GL_ATI_separate_stencil
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_clip_volume_hint
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_draw_buffers2
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture_array
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_s3tc
GL_EXT_texture_env_add
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array_bgra
GL_IBM_rasterpos_clip
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_light_max_exponent
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod

Original Prusa i3 MK3 / Firmware 3.9.3
stock model

When slicing on no scaling there is no issues. But after scaling model to 105% and infill 90% with Brim and vertical shell perimeters 7,  when the process of slicing is running the preview displays and then disappears when the process finishes(mouse pointer from a watch to arrow).

To reproduce the problem
1. Scale the model to any % ie. 105
2. Change infill to 60 -100%.
3. Include Brim.
4. 0.20mm Speed
5. Generic ABS
6. No Support.
7. Vertical Shells Perimeters = 7

[image]

[Shimano_BB_Socket_Small.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/6248061/Shimano_BB_Socket_Small.3mf.zip)

Thanks
Regards 
Ben

combolek, 2021-04-02:
Maybe this is Mac specific. It seems to work fine on Linux and Windows.
[image]

Commander-Firestrike, 2021-04-08:
Same thing happening to me on Windows. Without even changing any print settings, clicking "Slice Now" or going to the "Preview" view starts the slice, which is visible for a second or two, before disappearing. Enabling "Brim" makes the model visible - but without the brim added. Clicking "Slice now" again re-slices the model with the brim added, but once again it appears for a second then disappears, although the split second preview does show a brim.

hisashime, 2021-04-08:
My problem solved by using this unpublish version from #5566 
https://github.com/prusa3d/PrusaSlicer/issues/5566#issuecomment-781890033
https://www.dropbox.com/s/q4i4gf1fn2cerin/PrusaSlicer-2.3.0%2B278-universal-gfbd8d57ad-202102190006.dmg?dl=0

hisashime, 2021-04-15:
The latest version of PrusaSlicer 2.3.1-rc have solve the problems. Thank you.

closed 2021-04-15 by hisashime