prusa3d/PrusaSlicer #6305: Feature Request: Local Axis for Rotation/Scale/Move controls

hantonyan, 2021-04-02:
### Version
2.3.0+Win64

### Operating system type + version
Windows10

### 3D printer brand / version + firmware version (if known)
Prusa I3 MKS+

### Behavior
Rotation and Scale controls are very inconvenient, because the axis are always GLOBAL/WORLD and are not based on the orientation of the model. Instead a more often it is much more convenient is when the axis are LOCAL/RELATIVE, i.e. are based on the object itself and if you for example rotate the object, then the axis of rotation change with the object itself. Take a look for example at Unreal Engine editor controls, MAYA, or any other modeling tool.

The problem with the current implementation of PrusaSlicer (i.e. Global axis) is that if you have an object that is rotated in more than one axis, and if you want to rotate it for example by 30% to the right then you will have to again rotate in several axis (god knows how many degrees in each of the axis) while with Local axis approach you would have to rotate just one axis by exactly 30%.
I had a Dragon model and wanted to create modifier boxes with a higher infill covering the two wings of the dragon. The wings were not orthogonal to any of the axis, so to put the modifier boxes so that only wings get covered was very difficult, I had to rotate, scale, rotate, scale and it was very hard as there was no option to make axis Local to the object (modifier box).

See following examples of other tools, https://youtu.be/BvsN5GzxoHo?t=97, https://youtu.be/ralshA7aWag

_Is this a new feature request?_
YES

neophyl, 2021-04-02:
Have you tried switching the coordinates to Local on the Object manipulation panel for that object ?

hantonyan, 2021-04-02:
I just checked it and couldn't find any practical difference related with move/rotate/scale operations. It seems the only thing that Local/Global manipulation affects is just the numbers on the right side of the screen under the row labled as "Size" and I don't understand what kind of practical value do those numbers provide.

closed 2026-09-29 by prusaslicer-bot[bot]