prusa3d/PrusaSlicer #6306: Vase mode fillet bottom edge

mandy2tom, 2021-04-02:
Let’s keep it simple In VaseMode we need extra strength between the bottom and the side a simple chamfer or fillet with solve it

rtyr, 2021-04-07:
I am not sure why is it needed, but it would be not possible to print chamfered layers in the vase mode anyway. We may possibly implement vase mode per layer range in the future (combining of normal and vase mode like mentioned in https://github.com/prusa3d/PrusaSlicer/issues/798). Closing.

closed 2021-04-07 by rtyr

mandy2tom, 2021-04-07:
Almost every vase I’ve printed leaks water between the sidewalk and the bottom Requiring a few drops of acrylic glue for PTEG or some type of clear paint which leaves some discoloration. Basically just looking for a water tight VaseMode And I’ve also found that  Cura leaves no seam Anyway love the work you’ve done with PrusaSlicer it’s amazing and will continue to buy and support products from prusa To encourage the wonderful work That’s happening at Prusa!

bubnikv, 2021-04-07:
We have the smooth transition between the bottom and the first turn of the spiral in PrusaSlicer 2.3, but only for relative extruder mode.