prusa3d/PrusaSlicer #6309: Super laggy performance since a recent Windows or Nvidia Update

gopgnz, 2021-04-03:
### Version
Version 2.3.0+win64

### Operating system type + version
Windows 10
AMD Ryzen Threadripper 1950X 16-Core Processor  3.40 GHz
RAM 32,0 GB
Nvidia RTX 3090

### 3D printer brand / version + firmware version (if known)
Creality CR10-S

### Behavior

Since a recent Windows 10 update or maybe it was an Nvidia driver update, I'm experiencing really bad lagging using Prusa Slicer.. not just occasional stutters but it's constant. The main tab with the 3D models seems to run fine (at least navigating and rotating the 3D view is not affected) but within any of the settings tabs everything is slow and laggy. Switching between tabs takes 1-2 sec, clicking onto any setting takes the same time as well. While in sliced view scrolling the layer bar lags heavily as well. Simply moving the whole Prusa Slicer window (when not in maximized view) lags heavily. It's a really weird issue and having experienced this type of behaviour years ago in a different app it feels like a GPU issue but I'm not sure. This continued even after restarting the whole system.

#### Project File (.3MF) where problem occurs

gopgnz, 2021-08-02:
I have tried whatever I could, including reinstalling the app and installing recent Nvidia drivers but the issue is still there. My operating system is set to the native 4k resolution of my screen with 150% scaling which is recommended in the screen settings. Very curious how noone else is having such issues. Also I have none of this happening in any other software I use.

The only "solution" so far has been to set custom DPI settings by right clicking PrusaSlicer and choosing that Windows, instead of the app handles the DPI. This resulted in a lower resolution within the whole app which is blurry and smoothed out... however the lags are gone then.

ssill2, 2021-08-02:
Not sure.  I have dual sli 2080TI's with latest windows updates and latest nvidia drivers and I'm not seeing any kind of issues.

I tried reproducing your issue in 2.3.3 and the master build as built about 2 hours ago.  are you using any special features of the nvidia software?

ssill2, 2021-08-02:
I am on i9 16core cpu though, so that's definitely different.  
I'm running 64GB of ram and 2TB NVMe storage.  
I have this motherboard
https://rog.asus.com/us/motherboards/rog-strix/rog-strix-x299-e-gaming-model/

I mean it certainly could be something with the drivers for the 3090 but I've not been lucky enough to get one of those puppies.
I'll have to be content with my setup
[image]

gopgnz, 2021-08-02:
Not that I know of. I am using Nvidia's studio drivers, but I had the issue even before I started using that.
I just did another search and stumbled on this thread that seems to be very similar on my end. When I scale the app window down things seem to work just fine and I can switch between tabs easily and smoothly. Also, dragging the whole app window around is smooth. Once I scale the window at around 3/4 of my whole screen things get much slower, laggier. Maximizing the whole window makes it the slowest. The only thing that remains smooth in the app is rotating the actual 3D view. But anything that has to do with the overall UI gets crazy slow, taking 1-2 sec to even switch between settings tabs.

https://github.com/prusa3d/PrusaSlicer/issues/4026

ssill2, 2021-08-02:
if I maximize to the full screen on 4k display there's about a .5-.75sec delay switching from plater view to one of the tabs.  This delay is noticeable in 2.3.3, but non existent in latest master build from earlier today.   sadly I don't have a 3090 to test :)  I'm using the gaming drivers for the 2080's so I'm not sure if that's making a diff.

[image]

gopgnz, 2021-08-02:
>  This delay is noticeable in 2.3.3, but non existent in latest master build from earlier today.

Interesting, thank your for testing all this! How can I get that master build that you mentioned? I only see 2.3.3 being available.
Btw. I just recorded this video and to be honest this is still the smoothest I have seen it (probably because no other apps are running in the back and because I disabled my deskptop icons for the recording):
https://www.youtube.com/watch?v=Uoq4pcVdOpY

ssill2, 2021-08-03:
You'll have to build the master branch from source if you want to try it.  I build it nearly every day, or when I see changes committed.   I expect they'll be releasing an alpha soon'ish.  I've been running it for a while now since I wanted the MMU-painting feature.

ssill2, 2021-08-03:
I just did a similar test to what you did in the vid.  2.3.3 has some jumpiness when moving the window around, but 2.4.0 master build does not.  so something was fixed in master that's not in stable.

gopgnz, 2021-08-03:
Thanks a bunch for the insights! Then I will eagerly await the new release.

ssill2, 2021-08-03:
it would be nice if there was a place to pull nightly builds from.

ssill2, 2021-08-03:
if you hit me up in some private fashion I can probably share a build with you.

gopgnz, 2021-08-03:
Are you on the Prusa discord server? If yes, tell me your nickname there please

ssill2, 2021-08-03:
if you log on to the prusa forums and hit me up with private message, same username.

gopgnz, 2021-08-03:
Logged on there and found you but seems like I can't send DMs yet. Probably a minimum time needs to be spent there or minimum posts made.

ssill2, 2021-08-03:
maybe I can send you one.  same username?

gopgnz, 2021-08-03:
No, it's Andreas Mass

ssill2, 2021-08-03:
if you mouse over the little picture at the top right of forum screen it should say your username first thing.

gopgnz, 2021-08-03:
As I mentioned above, it's "Andreas Mass" - it seems like the forums created the account automatically based on my normal prusaprinters.org profile where I upload my designs.

ssill2, 2021-08-03:
Alright, stand by.

ssill2, 2021-08-03:
just sent you a private message there.  Can you see it?

gopgnz, 2021-08-03:
I can but it still won't let me answer, so stupid. Can you please just send me a mail to "hello@andreas-mass.com"? I will just delete these messages here later to not leave unnecessary cross references and contact info. Very much appreciate your time and effort to help out.

ssill2, 2021-08-03:
alright.

ssill2, 2021-08-03:
Just to close the loop on this for everyone following.  A build of master from yesterday made his issue go away.  

> Thanks a bunch for the insights! Then I will eagerly await the new release.

gopgnz, 2021-08-03:
> 
> 
> Just to close the loop on this for everyone following. A build of master from yesterday made his issue go away.
> 
> > Thanks a bunch for the insights! Then I will eagerly await the new release.

Exactly, he graciously helped me out and the new master build seems to use a slightly different looking UI (at least when it comes to how the main settings tabs look). And it performs really well even with 4k resolution!

bubnikv, 2021-08-04:
@YuSanka rewrote the "Tabbed notebook" control to her own implementation, as we were not able to convince the Win32 "tabbed notebook" to render nicely in dark mode. We are happy that not knowing we have solved two issues.

gopgnz, 2021-08-04:
> @YuSanka rewrote the "Tabbed notebook" control to her own implementation, as we were not able to convince the Win32 "tabbed notebook" to render nicely in dark mode. We are happy that not knowing we have solved two issues.

I should participate more often via GitHub by ways of sharing bugs and performance issues... not only because I was helped but also because I could take a glimpse of what is going on backstage as well as being able to confirm solved issues. I was kinda surprised that only few people have experienced or rather reported the same issue I was facing. 2.3.3 is so laggy for me running on 4k screen and at this point surely many other people use screens of this size, too.

bubnikv, 2021-08-11:
@gopgnz 

> I should participate more often via GitHub by ways of sharing bugs and performance issues... not only because I was helped but also because I could take a glimpse of what is going on backstage as well as being able to confirm solved issues. I was kinda surprised that only few people have experienced or rather reported the same issue I was facing. 2.3.3 is so laggy/stuttery for me running on 4k screen and at this point surely many other people use screens of this size, too.

Win32 controls are quite expensive and Microsoft does not help either, they seem to hope that we developers will replace Win32 controls with the "new" GUI toolkit that Microsoft delivers, however the "new" GUI toolkit changes too often and no native (C++) has really drawn any attention of the developers yet.

Speed of Win32 controls depends heavily on the number of pixels they occupy. Thus Win32 on a 4k display is expensive and slow.

We hope @YuSanka fixed (by accident) your issue with 4652733201f446f062e37574d4a44f80c1541592 by rewriting the tabbed notebook control. The fix will be released with PrusaSlicer 2.4.1-alpha1. Closing.

closed 2021-08-11 by bubnikv

gopgnz, 2021-08-11:
> Win32 controls are quite expensive and Microsoft does not help either, they seem to hope that we developers will replace Win32 controls with the "new" GUI toolkit that Microsoft delivers, however the "new" GUI toolkit changes too often and no native (C++) has really drawn any attention of the developers yet.
> 
> Speed of Win32 controls depends heavily on the number of pixels they occupy. Thus Win32 on a 4k display is expensive and slow.
> 
> We hope @YuSanka fixed (by accident) your issue with [4652733](https://github.com/prusa3d/PrusaSlicer/commit/4652733201f446f062e37574d4a44f80c1541592) by rewriting the tabbed notebook control. The fix will be released with PrusaSlicer 2.4.1-alpha1. Closing.

Fascinating, didn't realize how much devs are at the mercy of what Microsoft provides or rather doesn't provide :D ...especially weird if you consider it's 2021. I'm running AAA game titles in 4k on ultra settings with my 3090. Meanwhile a simple 2D UI of a Slicer is stuttering while there's not even any realy computation/calculation taking place. Thanks for the insights!