prusa3d/PrusaSlicer #6313: Segmentation fault after slicing some STL

fabienInizan, 2021-04-04:
### Version
2.3.0

### Operating system type + version
Archlinux, package  community/prusa-slicer 2.3.0-1

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3

### Behavior
I have an STL that creates a segmentation fault just after slicing.
I cannot reproduce this behavior with other files.

To reproduce:
 - Printer profile `Original Prusa i3 MK3`
 - Filament: `Prusament PLA`
 - Print Settings: `0.20mm SPEED`
 
Load the file and click the `Slice now` button. The part is sliced, renders, and a couple seconds later the software crashes with a segmentation fault.

#### Project File (.3MF) where problem occurs
[Lift.zip](https://github.com/prusa3d/PrusaSlicer/files/6254416/Lift.zip)

combolek, 2021-04-05:
I cannot reproduce it. Can you try with PrusaSlicer-2.3.0+linux-x64-GTK3-202101111352.AppImage ?
[image]

joeybronzoni, 2021-04-06:
Hey @combolek, I am dealing with the same issue. Running Arch Linux with slicer Version 2.3.0 and the crashes are kind of random but mostly when I slice. Doesn't matter which model from the simple cube to a complex whatever. I've tried some of the suggestions out there like running prusa-slicer from the terminal or installing the latest .git version from the AUR(which fails for an entirely different reason) and also tried the AppImage and still get the same results. 
`[1]    160310 segmentation fault (core dumped)  ./PrusaSlicer-2.3.0+linux-x64-202101111322.AppImage `
or: 
`[1]    165994 segmentation fault (core dumped)  prusa-slicer`
Typically it might finish the slice but then once touch anything from there it crashes. Sometimes it works for a short amount of time. I doubt its anything like a lack of resources just to cross that off the list. 
`Mem:            62Gi       4.9Gi        50Gi       1.1Gi       7.7Gi        56Gi
AMD ATI Radeon RX 5600 OEM/5600 XT / 5700/5700 XT
 NVIDIA GeForce RTX 2070 SUPER
And a sh*te ton of cores`

Also here's some stack trace from the journal logs: (there's a long list let me know if there's anything else I can say/do/or show.)
```
systemd-coredump[166714]: Process 165994 (slic3r_main) of user 1000 dumped core.

                                                    Stack trace of thread 165994:
                                                    #0  0x00007ff12b3838e5 n/a (radeonsi_dri.so + 0xbe78e5)
                                                    #1  0x00007ff12adafd62 n/a (radeonsi_dri.so + 0x613d62)
                                                    #2  0x00007ff12a97ed09 n/a (radeonsi_dri.so + 0x1e2d09)
                                                    #3  0x00007ff12a97bff8 n/a (radeonsi_dri.so + 0x1dfff8)
                                                    #4  0x00007ff12a97ca7a n/a (radeonsi_dri.so + 0x1e0a7a)
                                                    #5  0x00007ff12aa3874c n/a (radeonsi_dri.so + 0x29c74c)
                                                    #6  0x00007ff12aa3b4fa n/a (radeonsi_dri.so + 0x29f4fa)
                                                    #7  0x000055e0c4b1de71 n/a (prusa-slicer + 0xb23e71)
                                                    #8  0x000055e0c4b28f4e n/a (prusa-slicer + 0xb2ef4e)
                                                    #9  0x000055e0c4a9f8ee n/a (prusa-slicer + 0xaa58ee)
                                                    #10 0x00007ff142b35dde _ZN12wxEvtHandler23ProcessEventIfMatchesIdERK21wxEventTableEntryBasePS_R7wxEvent (libwx_baseu-3.0.so.0 + 0x1dedde)
                                                    #11 0x00007ff142b3615b _ZN12wxEvtHandler23SearchDynamicEventTableER7wxEvent (libwx_baseu-3.0.so.0 + 0x1df15b)
                                                    #12 0x00007ff142b361f1 _ZN12wxEvtHandler11TryHereOnlyER7wxEvent (libwx_baseu-3.0.so.0 + 0x1df1f1)
                                                    #13 0x00007ff142b362ac _ZN12wxEvtHandler19ProcessEventLocallyER7wxEvent (libwx_baseu-3.0.so.0 + 0x1df2ac)
                                                    #14 0x00007ff142b36382 _ZN12wxEvtHandler12ProcessEventER7wxEvent (libwx_baseu-3.0.so.0 + 0x1df382)
                                                    #15 0x000055e0c4b668c3 n/a (prusa-slicer + 0xb6c8c3)
                                                    #16 0x000055e0c4b4b1ab n/a (prusa-slicer + 0xb511ab)
                                                    #17 0x000055e0c4b49c54 n/a (prusa-slicer + 0xb4fc54)
                                                    #18 0x00007ff142b35dde _ZN12wxEvtHandler23ProcessEventIfMatchesIdERK21wxEventTableEntryBasePS_R7wxEvent (libwx_baseu-3.0.so.0 + 0x1dedde)
                                                    #19 0x00007ff142b3615b _ZN12wxEvtHandler23SearchDynamicEventTableER7wxEvent (libwx_baseu-3.0.so.0 + 0x1df15b)
                                                    #20 0x00007ff142b361f1 _ZN12wxEvtHandler11TryHereOnlyER7wxEvent (libwx_baseu-3.0.so.0 + 0x1df1f1)
                                                    #21 0x00007ff142b362ac _ZN12wxEvtHandler19ProcessEventLocallyER7wxEvent (libwx_baseu-3.0.so.0 + 0x1df2ac)
                                                    #22 0x00007ff142b36382 _ZN12wxEvtHandler12ProcessEventER7wxEvent (libwx_baseu-3.0.so.0 + 0x1df382)

```

combolek, 2021-04-06:
This stack trace suggests that it's crashing in the AMD GPU driver. If that's indeed the case, you may be able to slice using the command line like this:
```
prusa-slicer --slice Lift.3mf
```

fabienInizan, 2021-04-06:
I confirm I have the same behavior than @joeybronzoni, with core dump pointing to radeonsi_dri.so. The command line slicing works with no issue.

@combolek If this is unrelated to prusa-slicer please let me know and I'll close the issue.

joeybronzoni, 2021-04-06:
Ahh I see, I haven't been using this slicer long but that makes sense and works just fine.  (typical mesa headaches with this gpu). I will try installing some more specific drivers and I'll let ya'll know if it helps. But yea that's works -as God intended... all things in the cli.... Thanks @combolek.

combolek, 2021-04-06:
> @combolek If this is unrelated to prusa-slicer please let me know and I'll close the issue.

I don't know for sure

angel6700, 2021-04-06:
Hello,

I have the same issue here. Similar system:
Archlinux, radeon card (RX590).

I suspect it is not related to prusa-slicer, but to mesa, that was updated recently to version 21.

The only real workaround is to define first:  LIBGL_ALWAYS_SOFTWARE=true

joeybronzoni, 2021-04-06:
Thanks @angel6700. That works. A little sluggish but still works. Much appreciated.

angel6700, 2021-04-07:
This issue in mesa seems related:
https://gitlab.freedesktop.org/mesa/mesa/-/issues/4534

apparently it has been already been fixed, so you have three options now:
1- Wait for a new stable release of mesa with the patched included and archlinux package to be build against it.
2- Build mesa from source git, and test it (it is easy from aur, as there is a mesa-git aur pkgbuild)  I did it and it works (beware of other unexpected bugs in other places, as this is the development version).
3. Downgrade mesa.

I think the issue could be closed as it is not related to PrusaSlicer.

fabienInizan, 2021-04-07:
Downgrading mesa fixed the issue, thanks @angel6700 for investigating the root cause.

Closing as unrelated.

closed 2021-04-07 by fabienInizan