prusa3d/PrusaSlicer #6315: underflow when printing 2 items

Daniel599, 2021-04-04:
### Version
Version 2.3.0+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa MINI firmware 4.3.1+2759, stock model

### Behavior
Using the generated GCODE by PrusaSlicer to print 2 earrings side by side, the right earring's support doesn't print well,
seems like underflow.
* _Expected Results_
[image]

* _Actual Results_
[image]
[image]
[image]

_Is this a new feature request?_ No.

#### Project File (.3MF) where problem occurs

[50mm-KoiFish-earings-mxu9560.zip](https://github.com/prusa3d/PrusaSlicer/files/6255137/50mm-KoiFish-earings-mxu9560.zip)
[50mm-KoiFish-earings-mxu9560_0.05mm_PLA_MINI_5h7m.zip](https://github.com/prusa3d/PrusaSlicer/files/6255144/50mm-KoiFish-earings-mxu9560_0.05mm_PLA_MINI_5h7m.zip)
STL from thingverse: https://www.thingiverse.com/thing:3534703

P.S sorry about the title, I don't know how to describe this problem well.

rtyr, 2021-04-07:
I was not able to reproduce it with your gcode. You may try to increase extrusion width for support material, it would be also better to place those instances closer to each other to reduce travel distance (and oozing during that travel).

Daniel599, 2021-04-07:
I`ll try to print them closer and see if it helps

Daniel599, 2021-05-07:
If I print them closer it doesn't happen, so I`m closing this, thanks.

closed 2021-05-07 by Daniel599