prusa3d/PrusaSlicer #6317: Permit to change font size

henri9813, 2021-04-05:
### Version

2.3.0
### Operating system type + version

MacOS Big Sur

### 3D printer brand / version + firmware version (if known)
Prusa I3 MK3S+ 

### Behavior
I could like to change the font size, it's very small for me./

It's a feature request

[image]

I had to be very near from the screen to see correctly 
#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2021-04-07:
PrusaSlicer uses the font size configured in your system. Shouldn't you
increase your system font size?

redvelociraptor, 2021-08-22:
I'm working on MacOS Mojave, also using PrusaSlicer 2.3.0. I use a 4k screen on my Mac, the fonts in PrusaSlicer are very small at this resolution.

MacOS doesn't have the concept of system font size @bubnikv. The only way to increase an app's font size is either reduce screen resolution, or if the individual app provides the option to increase font size.

In the screenshot, I have the Safari set to never set display font to smaller than 18. This Github page open on the right, Prusa Slice size on the bottom. The only app I use for 3d work that gets this "right" is Blender. It allows you to change the resolution scale of the entire UI.
[image]

bubnikv, 2021-08-23:
Let's wait for more feedback. I discussed this topic with our tester @FidelCapo who is an avid OSX / Apple user. Allowing an application font size under OSX is not a usual thing to do. We suppose the way to go is to change your screen scaling. Why don't you do that?

henri9813, 2021-08-23:
Hello,

Because i don't need zoom for the rest of my apps :-) 

Sorry, this request is pretty unusual...

Best regards,
Henri

bubnikv, 2021-08-23:
> Because i don't need zoom for the rest of my apps :-)

What is so special about PrusaSlicer? Why does the default text size not work for you in PrusaSlicer if it works for you in other applications (possibly with the exception of the web browser)?

henri9813, 2021-08-25:
I zoom in webbrowser, and in my opinion, input are small and "text too". 

I just an improvement or en enhancement to improve my life in midniht when i try to print with some specials settings :D

edwardhorsford, 2021-09-10:
A big +1 from me. I'm also on a 4k display, and the gCode text (just in PrusaSlicer) is really hard to read.

The UI text is ok, but the gCode font specifically is much smaller. Screenshot below.
[image]

I don't know if it's intentional, but it's significantly smaller than 'default'. Could it be increased to have more of a parity in size?

jtjohnson1, 2021-11-24:
+1 for me too.  It seems this is a universal issue as the Windows folks seems to have a slider for font size:  https://github.com/prusa3d/PrusaSlicer/issues/5121  and they are asking for additional features for it.

@bubnikv I think it's been made pretty clear that changing system font sizes is not a good solution.  Honestly, I don't really care about your personal feelings on how to make my everyday life easier:  

> Frankly I think one shall increase font size of the system

This is a pretty inconsiderate comment if you're going to support Mac as a platform.  There's an obvious need here and your suggestions will not work in the eyes of the user community.  Windows users already have this feature and they are asking for updates to it.  Mac is different, we're asking for a feature to be added.  If this is going to be a "Wont Do", the please let us know so we can either move to a different developer or find a replacement to PrusaSlicer.

timperlabs, 2022-03-11:
+1

I've seen many projects resist the ability to change font size and every single time, the user community wants it.  Not being to see clearly is irritating and leads to great annoyance.   This is of far greater impact than debates about, for example, light on dark vs dark on light.   Even Apple expects apps to support this (see below).  

[https://support.apple.com/guide/mac-help/make-it-easier-to-see-whats-on-the-screen-mchld786f2cd/mac](url)
> Make text bigger on your Mac
> Increase the text size of emails in Mail: In Mail, choose Mail > Preferences, click Fonts & Colors, click Select next to “Message font,” then select a font size in the Fonts window.
> Increase the text size of messages in Messages: In Messages, choose Messages > Preferences, click General, then move the “Text size” slider to the right.
> Increase the text size in other apps: In many apps, you can press Command-Plus (+) or Command-Minus (–) to adjust text size. If that doesn’t work, check the app’s preferences.

tempire, 2022-04-08:
Indeed; you folks should have at least one mac user on your dev team for a wider understanding of UX design. I want to like Prusaslicer, but the UI is a strange mix of mid-2000s web, mid-90s jwt, and x11 windowmaker.

strugatzki, 2022-05-09:
+1 from me (although in all fairness e.g. the standard text size when quicklooking .ini files in the OS is comparably small)
you don't have to implement a menu, but if you could point us to a file we can edit or maybe could tell us what we could put into the PrusaSlicer.ini file, that would be great!

workaround: use zooming in the macos accessibility options (enable zooming by pressing control and scrolling, works like a charm)

Plumbob, 2022-09-05:
If modifying the GUI or preferences pane is a problem, the Mac universal Mac keyboard shortcuts of  CMD - (decrease font size) or CMD + (increase font size) would be greatly appreciated.

kermitxc, 2022-11-07:
+100 on this - it is very hard to read the fonts on a 4k display

mrelwood, 2022-11-21:
This issue at first stopped me from even properly trying PrusaSlicer. I now noticed that I can get better quality prints with Prusa, so I'd really like to use it. But all the fonts in the application are smaller than on any other application or the system, even on just 16" MBP's own display. The issue is glaring enough that if the dev team had had a Mac guy, it would've been fixed before first release.

PLEASE add a scaling option, or even just increase all app fonts on the Mac version to for example 150%. I can't see how anyone would complain about it.

[image]

edwardhorsford, 2022-11-21:
Personally, I feel like the first priority should be a better default font size that matches the apparent size of other Mac UI elements.

mobigroup, 2023-01-13:
Unfortunately, it still looks terrible for build 2.5.0+MacOS-arm64:

[image]

I don't believe PrusaSlicer developers can easy read all the text on the screenshot above... but how could users read that?!! By the way, Cura slicer allows to change all the fonts in configuration file.

candera, 2023-07-06:
+1. If you're over 45 or so, you probably know that this is a common issue in many apps. Those that do it well are much easier for many older (and otherwise vision-impared) users to use.

This is an accessibility issue.

lukasmatena, 2023-10-23:
Implemented in [PrusaSlicer 2.7.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-alpha1). Closing.

closed 2023-10-23 by lukasmatena