prusa3d/PrusaSlicer #6320: Won't build with cmake+ninja (parallelization issue ?)

Salamandar, 2021-04-06:
### Version
git master.

### Operating system type + version
Archlinux (AUR package prusa-slicer-git)

Here's the build issue :  (sorry, my OS is in french)
```
[273/392] Building CXX object src/libslic3r/CMakeFiles/libslic3r.dir/OpenVDBUtils.cpp.o
FAILED: src/libslic3r/CMakeFiles/libslic3r.dir/OpenVDBUtils.cpp.o 
/usr/bin/c++ -DBOOST_ALL_NO_LIB -DBOOST_ATOMIC_DYN_LINK -DBOOST_CHRONO_DYN_LINK -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_LOCALE_DYN_LINK -DBOOST_LOG_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DLIBNEST2D_GEOMETRIES_clipper -DLIBNEST2D_OPTIMIZER_nlopt -DLIBNEST2D_STATIC -DLIBNEST2D_THREADING_tbb -DSLIC3R_GUI -DTBB_USE_CAPTURED_EXCEPTION=0 -DUNICODE -DUSE_TBB -DWXINTL_NO_GETTEXT_MACRO -D_UNICODE -DwxNO_UNSAFE_WXSTRING_CONV -DwxUSE_UNICODE -I/usr/include/dbus-1.0 -I/usr/lib/dbus-1.0/include -I../src -Isrc/platform -I../src/clipper -I../src/libslic3r -Isrc/libslic3r -I../src/libnest2d/include -I../src/miniz -I../src/glu-libtess/include -I../src/qhull/src -isystem /usr/include/eigen3 -isystem ../src/libigl -isystem /usr/include/OpenEXR -D_FORTIFY_SOURCE=2 -march=x86-64 -mtune=generic -O2 -pipe -fstack-protector-strong -fno-plt -fext-numeric-literals -Wall -Wno-reorder -O3 -DNDEBUG -fPIC -fsigned-char -Werror=return-type -Wno-ignored-attributes -Wno-unknown-pragmas -pthread -DOPENVDB_ABI_VERSION_NUMBER=8 -std=gnu++17 -Winvalid-pch -include /home/piedallu/.cache/yay/prusa-slicer-git/src/PrusaSlicer/build/src/libslic3r/CMakeFiles/libslic3r.dir/cmake_pch.hxx -MD -MT src/libslic3r/CMakeFiles/libslic3r.dir/OpenVDBUtils.cpp.o -MF src/libslic3r/CMakeFiles/libslic3r.dir/OpenVDBUtils.cpp.o.d -o src/libslic3r/CMakeFiles/libslic3r.dir/OpenVDBUtils.cpp.o -c ../src/libslic3r/OpenVDBUtils.cpp
{entrée standard}: Messages de l'assembleur:
{entrée standard}:279000: Avertissement: fin du fichier n'est pas à la fin de la ligne; nouvelle ligne insérée
{entrée standard}:280432: Erreur: pas de telle instruction: « ad »
{entrée standard}: Erreur: ouverture CFI à la fin du fichier; directive .cfi_endproc manquante
c++: erreur fatale: le signal Complété a terminé le programme cc1plus
compilation terminée.
```

If I go into the build directory and manually run this command, it runs fine.

I suspect a parallelization issue during the build. Ninja being more performant than Make, it may reveal some issues in CMake scripting.

l29ah, 2021-05-09:
I suspect that you run out of RAM, check your `dmesg` for OOM events.

bubnikv, 2021-09-25:
> I suspect that you run out of RAM, check your dmesg for OOM events.

That may be.

closed 2021-09-25 by bubnikv