prusa3d/PrusaSlicer #6321: color change not working

acceleratedhgts, 2021-04-06:
On the prusaslicer 2.3.0 on my mac mini with M1 chip the color change does not work, I can set it but it does not stop printing, just prints the entire print in 1 color. I went back to version 2.2 and the color change works perfect. Is there a fix or a work around for this to work on 2.3.0? i want those other features like paint on supports etc.

rtyr, 2021-04-07:
Can you please provide your 3mf and gcode?

acceleratedhgts, 2021-04-07:
How do i get that? i will attach a gcode of a model that has color change
but it doesnt work. Im using a creality ender 3 pro.

rtyr, 2021-04-07:
File - Save project as...

[image]

You have to zip it first (or change file extension).
Also, it is unfortunately not possible to upload files via e-mail.

pmjdebruijn, 2021-04-08:
@acceleratedhgts Are you running stock firmware on your Ender 3 Pro? If so, which exact firmware version?

This may be a firmware limitation, if so, we can disable the feature, so we don't misrepresent the (stock) printer's capabilities...

rtyr, 2021-05-13:
No response, likely not PrusaSlicer related issue. Closing.

closed 2021-05-13 by rtyr