prusa3d/PrusaSlicer #6322: macOS - Finder - icon preview

GodricCZ, 2021-04-07:
### Version
PrusaSlicer 2.3.0
### Operating system type + version
MacOS Big Sur - ver 11.2.3
Macbook Air (M1,2020)
### Behavior

Ve Windows standardně v Průzkumníkovi fungoval přímo náhled (ikona) 3mf souboru, kde bylo jasně v ikoně a v náhledu vidět, co je vše obsahem 3mf souboru. V macOS toto zobrazení není možné. Je tedy možné implementovat místo zobrazení ikony PrusaSliceru přímo ikonu náhledu 3mf souboru?

bubnikv, 2021-04-07:
Nevíme. Víte o aplikaci která to na OSX dělá?

GodricCZ, 2021-04-07:
Bohužel netuším.. :/ přešel jsem z dlouhodobého užívání Windows na macOS a v něm se divím, že to není vidět.... a je to škoda, protože je to dobrá funkce. STL soubory jdou standardně vidět... Jinak jestli se nepletu, tak Windows mě 3mf soubor nezobrazí v Průzkumníkovi dokud nenainstaluji např. PrusaSlicer... tedy Windows podle mě využívá knihovnu zobrazení z Prusasliceru

devrosx, 2021-04-09:
Dělá to třeba **pleasant3d**  http://www.pleasantsoftware.com/developer//index.shtml ale jen u STL souborů, není to ale podporováno systémem, jako třeba obrazové formáty kde se to dělá automaticky, ale přes vložení obázku do ikony...

GodricCZ, 2021-04-11:
STL soubory jsou normálně zobrazované.. u nich není problém

FidelCapo, 2021-04-14:
macOS umí nativně zobrazovat STL soubory, dokonce ve 3d, ale neumí zobrazovat 3mf soubory. Nenašel jsem žádnou aplikaci která by zobrazování 3mf souborů umožňovala, ani způsob jak toho dosáhnout.

Win umožňuje zobrazování náhledů 3mf nativním převodem na obrázek, naopak zase neumožňuje zobrazování STL souborů, ovšem zobrazování STL souborů je možné instalací aplikace (například https://marlin3dprintertool.se/stl-thumbnail/) která podporuje jen Win.

Rozdíl v náhledech souborů je tedy především z důvodu rozdílnosti OS a jejich možnostech.

devrosx, 2021-04-14:
Jak jsem psal pleasant3d má vlastní generování STL náhledů (je to sice statický obrázek ale funguje to) a je opensource, takže podobným zpusobem by šel upravit kód PrusaSliceru aby generoval náhledy pro 3MF. Nebo počkat až apple 3MF začne podporovat v nativně :)

bubnikv, 2021-04-14:
> Jak jsem psal pleasant3d má vlastní generování STL náhledů (je to sice
statický obrázek ale funguje to) a je opensource, takže podobným zpusobem
by šel upravit kód PrusaSliceru aby generoval náhledy pro 3MF. Nebo počkat
až apple 3MF začne podporovat v nativně :)

Tohle?
Pleasant Software: Pleasant3D
<http://www.pleasantsoftware.com/developer//pleasant3d/>

To vypadá jako aplikace, ne jako něco co by se integrovalo do finderu.
Nerozumím tomu.

devrosx, 2021-04-14:
tady jsou k tomu zdrojové soubory https://github.com/zaggo/Pleasant3D/tree/master/src/subprj (GCodeQuickLook a STLQuickLook)
používá to QuickLook knihovny... 
pokud tuto aplikaci nainstalujete máte preview na STL a GCODE soubory ve Finderu...

bubnikv, 2021-04-15:
aha, díky. Zkusíme.

DavidPhillipOster, 2023-02-04:
I've written an open-source command line tool that extracts the thumbnail from a PrusaSlicer gcode file and applies it as the Finder icon: https://github.com/DavidPhillipOster/gcodeToThumbnail

The important part is just the pair of lines:

            NSWorkspace *ws = [NSWorkspace sharedWorkspace];
            if (nil == img || ![ws setIcon:img forFile:[NSString stringWithUTF8String:s]options:0]) {…}

That is, it is a single function call, built into the standard library on macOS, to associate a 512x512 image as the icon of a file. Yes, that's Objective-C, but Objective-C method calls are directly convertable to C function calls.

bubnikv, 2023-02-04:
I see, so we may associate an image to a file we produce. Nice. How does
Finder process moving, renaming and deleting the files? Will the thumbnail
association break when a file is moved or renamed?

DavidPhillipOster, 2023-02-04:
The association doesn't break when the file is moved or renamed, as long as you do it  using the Finder. When the file is deleted, the image is gone. You can see this on macOS, by in Finder, selecting any file or folder and doing a Get Info (⌘I) on it. In the Get Info window, you can select the icon in the top left, and cut, copy, paste, or delete it. (Deleting it causes it to revert to the default icon for that file extension) Prusa uses this to give the icon for the PrusaSlicer's folder a custom icon. The icon is actually stored in the file's meta-data. Use:

    ls -l@  theFile

to see the Apple specific metaData attributes that are stored in the file system with the data.  Use `man xattr` for more information.

Using the Finder's **compress** command creates a .zip file which, when unzipped, preservers the custom icon.

Since all of this is Apple-specific, copying a gcode file to an EXFAT or FAT filesystem strips off the custom icon.

bubnikv, 2023-02-04:
Thanks, David for the explanation.

Do you recommend to always store a thumbnail for 3MF? Or should we add a UI
switch to disable such behavior?
Do you recommend to always store a thumbnail for G-code? Even to SD card or
a thumb drive?

DavidPhillipOster, 2023-02-04:
I would recommend always storing a thumbnail in the 3MF - it will be small compared to the  total size of the .3MF file. Either JPEG or PNG (I'd prefer PNG) as documented in https://github.com/3MFConsortium/spec_core/blob/1.2.3/3MF%20Core%20Specification.md#61-thumbnail 

And, it looks like there already is a 256x356 .png  in the existing .3mf files, inside the .3mf, at 'Metadata/thumbnail.png'

Currently, there **is** a U.I. for turning the gcode thumbnail on or off, but it is so poorly documented that users will never find it. As shown in the screenshot in

https://github.com/DavidPhillipOster/gcodeToThumbnail/

if the **G-code thumbnails** line of the preferences is blank, then no thumbnail is created.

Given the discussion in https://github.com/prusa3d/PrusaSlicer/issues/4127 , having thumbnails on by default will cause issues for some users, but it would be nice if this were documented better.

(Also, I learn from that https://github.com/prusa3d/PrusaSlicer/issues/4127 that my code is only reading the first thumbnail. My code could be improved by scanning the gcode file for **all** the thumbnails and just keeping the largest one.)

In macOS Ventura, (13.1), the current macOS, in **Settings > Privacy & Security > Extensions > Quick Look** there is U.I. for enabling or disabling thumbnail generators like the one I'm planning to write.

I'm planning to write an extension (that could potentially ship bundled inside the PrusaSlicer app) that handles displaying in the Finder both the .gcode thumbnails (when present) and the .3mf thumbnails (which appear to be already present.)

I'll make sure to handle multiple, non-square thumbnails in gcode files, and that gcode thumbnails are optional.

## What I want from  you:  

Please improve the user-level documentation of the  **G-code thumbnails** line of the preferences.

Let users know that they might need a gcode thumbnail is they use a printer that displays those thumbnails, but that big thumbnails could cause some 8-bit octoprint installations to time out trying to skip large thumbnails.

DavidPhillipOster, 2023-02-05:
https://github.com/DavidPhillipOster/ThumbHost3mf  is an application host for a thumbnail generator extension. Ideally, the extension would just ship inside the PrusaSlicer app bundle, but I want to get early feedback before working on a pull request with that.

DavidPhillipOster, 2023-12-22:
I've enhanced https://github.com/DavidPhillipOster/ThumbHost3mf with support for binary gcode files and QOI format preview images (in addition to png and jpg) More discussion here: https://old.reddit.com/r/prusa3d/comments/18fn2xa/if_you_use_prusaslicer_on_mac_free_open_source/ including a favorable comment from https://old.reddit.com/user/josefprusa

lukasmatena, 2024-02-23:
@DavidPhillipOster Thanks, we will look into it. SPE-2173

closed 2026-09-29 by prusaslicer-bot[bot]