prusa3d/PrusaSlicer #6323: [Feature Request] PrusaViewer - solid background and three decimal places for tool position

fe60, 2021-04-07:
### Version
PrusaSlicer G-code Viewer-2.3.0+win64-UNKNOWN

### Operating system type + version
Windows 10 20H2

### Behavior
#### Current
I want to modify gcode in a text editor and watch the result in PrusaViewer. To identify the correct location in the file I often look for the tool position in the viewer and compare it with the values in the gcode file. This is harder than it has to be for two reasons:
1. The translucent background can make it hard to read the tool position (see second 1 of 181 in screenshot).
2. Looking for the coordinates shown in the viewer doesn't work well, as it is shown with two decimals in the viewer, while the values in the gcode come with a precision of three places. It gets even worse if the values have been rounded (compare position in viewer with selected row in editor).
[image]

#### Expected
Show tool position in PrusaViewer with three decimal places on a solid background.

_Is this a new feature request?_
Yes

combolek, 2021-04-07:
Tool position is already displayed with 3 decimal places in the latest 2.4 alpha:
[image]

fe60, 2025-01-24:
I don't think it's helpful to keep this open after such a long period of time.

closed 2025-01-24 by fe60