prusa3d/PrusaSlicer #6327: Object moves on X axis between 3D view and sliced view

Chrispizz, 2021-04-08:
### Version
Version 2.3.0

### Operating system type + version
Windows 10 - 64 bits

### 3D printer brand / version + firmware version (if known)
ANET A6

### Behavior
When I place my 3D object on the middle of the plate , after slicing the object moves 2.5cm to left on X axis 
After slicing , the object should be on the same place that I fixed it. A big object should be out of range after slicing due to this X move.

#### Project File (.3MF) where problem occurs
See bellow my pictures 3D view and Sliced view :
[Attached Files To Issue.zip](https://github.com/prusa3d/PrusaSlicer/files/6279064/Attached.Files.To.Issue.zip)

Chrispizz, 2021-04-08:
I think it will be good to see the config I use
I have exported them in the new file PrusaSlicer_config_bundle.ini , you will find it in the ZIP file
[Attached Files To Issue.zip](https://github.com/prusa3d/PrusaSlicer/files/6279276/Attached.Files.To.Issue.zip)

rtyr, 2021-04-08:
It is caused by your start g-code.

[image]

Chrispizz, 2021-04-08:
Wooooooo .... OK , if it's realy that , then I'm going to have a big problem.
In the G-code I use a G1 E5 too , this is to purge the head , and I don't see this draw in the sliced preview. 
If you are right , this center adjust by injected G-code G92 is here to correct the point (0,0) on my ANET A6 , I have to keep it to center my 3D print.
I was using CURA and my injected G-code was not interpreted in the sliced view. 
I imagine that the real solution will be to solve the offset X of my ANET A6 without A G92. 
Is there a way to not execute injected G-code in the sliced preview ?

neophyl, 2021-04-08:
Can’t you set the offset in the printer settings section, the area where you define the bed size.  Iirc (I’m not at my pc) but it also has an offset section too.

Chrispizz, 2021-04-08:
Thx Neophyl , I'm going to try this

lukasmatena, 2021-04-12:
Can this be closed? It appears that PrusaSlicer does what it is supposed to, no?

Chrispizz, 2021-04-12:
Yes ,sure , and thank you for the solution

Envoyé le :lundi 12 avril 2021 09:45
À : ***@***.***>
Cc : ***@***.***>; ***@***.***>

closed 2021-04-12 by rtyr