prusa3d/PrusaSlicer #6329: RepRapFirmware setting produces error

timnMK, 2021-04-08:
2.3.0 Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

Win10 Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

Wanhao duplicator i3 plus 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

Error in G-code output:
In the printer setting:
If I select G-code flavour as RepRap/Sprinter we expect and get this:
M107
M190 S80 ; set bed temperature and wait for it to be reached
M104 S250 ; set temperature

If I select the Gcode flavour as RepRapFirmware I get this
M107
M190 S80 ; set bed temperature and wait for it to be reached
G10 S250 ; set temperature

i.e. it uses G10 instead of M104 to set the filament temp and consequently does not extrude anything

rtyr, 2021-04-08:
G10 is correct command for RepRapFirmware flavor, but it apparently requires tool parameter as mentioned here https://github.com/prusa3d/PrusaSlicer/issues/5919

This should solve your problem https://github.com/prusa3d/PrusaSlicer/issues/5919#issuecomment-769645689

timnMK, 2021-04-08:
Ah - so RepRapFirmware doesn't follow the marlin G10 description.
 I'm pretty sure if I set RepRap/Sprinter then Prusa gives me a failed print in the initial layers... but I try that again
thanks

rtyr, 2021-04-08:
So the solution mentioned here https://github.com/prusa3d/PrusaSlicer/issues/5919#issuecomment-769645689 does not work for you (RepRapFirmware flavor + that line in the start g-code)?

Please provide your 3MF project file.

rtyr, 2021-04-08:
Looking at your printer model I assume that Wanhao i3 printer actually doesn't use RepRapFirmware, so there is no reason to use RRF gcode flavor.

Please provide your 3MF project file.

timnMK, 2021-04-08:
The Wanhao duplicator i3 plus, as badged by Balco, is my printer and Balco suggested I use RepRap (they weren't specific on Firmware or Sprinter).
All is well if I insert a custom code as suggested in #5919 and use RRF. 
This works for me:
**M109 S[first_layer_temperature] ; set temperature**

I might investigate the G10 again with P and the S variables.
Sorry I can't provide the propriety project file at the moment.
But thanks for the help - at least I'm printing some good quality models with Prusa

closed 2021-04-08 by rtyr