prusa3d/PrusaSlicer #6331: Alpha (from Master) support bug?

3Delights, 2021-04-09:
Alpha (Compiled from from Master? Used WIKI guide) version_2.4.0-alpha0-121-g558deca78

Windows 10 Pro 19042.906

MK3S+
Firmware: 3.9.3.3556

When adding supports to the attached model some of the support interfaces appear to be missing.  I am assuming this is a bug unless I am misunderstanding how they work.  This is using the new SNUG supports.

The Sliced model looks like this:
[image]

and the area in question like this:
[image]

I expected it to look like this:
[image]

Here is the Project file zipped:
[567 Column Test.zip](https://github.com/prusa3d/PrusaSlicer/files/6286853/567.Column.Test.zip)

3Delights, 2021-04-17:
It's actually worse than I thought as there are sections of supports missing along the length of the slot in the model:
[image]
Some of the supports seem to be over empty space!

bubnikv, 2021-04-19:
This is a known issue. The implementation is not ready.

I am going to close this issue for now.

Markus87, 2022-12-06:
Is this truly fixed? (`PrusaSlicer-2.5.0+win64-202209060714`)
Style: snug
[image]

Style: grid
[image]

I did not noticed this before printing, on the printer it looked like it did everything to not support its product. 😅 
But I am pretty new to this, it is as likely that it is a newbie mistake.