prusa3d/PrusaSlicer #6333: PrusaSlicer 2.3.0 slices holes into model.

Twilek-de, 2021-04-10:
### Version
2.3.0 win64

### Operating system type + version
Win 10 64

### 3D printer brand / version + firmware version (if known)
MK3S+

### Behavior
PrusSlicer slices holes into this model. Cura does not.

#### Project File (.3MF) where problem occurs
See Modelfile and Screenshot
[JaJa2.zip](https://github.com/prusa3d/PrusaSlicer/files/6289860/JaJa2.zip)
[image]

neophyl, 2021-04-10:
I downloaded your project and opened it in 2.3.0 on Windows 10.  Weirdly all your setting are NOT loading in.  Normally when loaded it will load in your printer/filament/print settings but none of those were updated from what I had previously selected.  This makes checking using the same settings problematic.

Regardless of that though the model has 680 errors reported as repaired.  It has many more than that though.  If you have detect thin wall enabled you get the missing perimeters you show in your picture.  If you turn off detect thin walls you get this missing areas properly sliced but the thin curved columns are then missing huge sections.  It's these areas that have most of the model errors.  However having detect thin walls enabled can cause its own problems.

Openscad is notorious for producing models with problems.  I exported the stl from PS and ran a check on it.  Blenders 3dprint tool reports, 293 intersecting faces, 979 zero faces, 532 zero edges.  At least its manifold.

I fixed it using the same tool and exported it.  Now if I slice it with Detect thin wall=OFF the model appears to be all there and slices correctly.  If I turn on detect thin walls the same bits get missed (as I say it can cause its own issues).

At the moment Cura's auto repair function handles common Openscad errors better.  I believe that is being looked at though.

Here is the fixed model I used to slice successfully..

[jaja2.zip](https://github.com/prusa3d/PrusaSlicer/files/6290079/jaja2.zip)

Twilek-de, 2021-04-10:
Thanks a lot for your effort. I just used Cura to print it. I wasn´t aware that OpenScad produces such problematic meshes. I found it interesting that PrusaSlicers displays the object alright with all outer surfaces intact and then gets completely off the rails when it tries to slice it. The 3mf file was something OpenScad itself exported so there weren´t probably any printer settings in it. Maybe the model can be used to improve Prusaslicer.

rtyr, 2021-04-10:
Handling of OpenScad models was already improved on master (9c5dd35), the fix will be included in the next PrusaSlicer release.

```
Fixing model import of models with nearly degenerate triangles, which
degenerate after transformation (translation or rotation).

There was certainly a bug in PrusaSlicer, however this bug was triggered
mostly by the ASCII STL files generated by OpenSCAD, likely due to insufficient
number of decimal digits when saving an ASCII STL, STL export should contain single precision floats

```

Provided model is sliced correctly on current master. Closing.

closed 2021-04-10 by rtyr