prusa3d/PrusaSlicer #6337: [Feature Request] BL-Touch Version of Creality's Ender-6 Printer

korzyn1, 2021-04-11:
### Version 2.4.0

Please add a BL-Touch Version of Creality's Ender-6 Printer.

Thank you

neophyl, 2021-04-11:
All the non prusa printers are submitted by users themselves.  Some guidelines can be found here https://github.com/prusa3d/PrusaSlicer/wiki/Vendor-bundles-and-updating-process

pmjdebruijn, 2021-04-12:
As far as I'm aware there's no Ender-6 with BLTouch from factory. Presumably you performed that upgrade yourself?

With the huge amount of Creality printers already, adding a BLTouch variant to each and every one of them, the printer list will become rather unwieldy, so I'm not inclined to go down that route. At least not without an elegant solution which currently doesn't seem possible.

@rtyr: unless you feel differently?

rtyr, 2021-04-12:
> With the huge amount of Creality printers already, adding a BLTouch variant to each and every one of them, the printer list will become rather unwieldy, so I'm not inclined to go down that route.

I agree. Closing.

closed 2021-04-12 by rtyr

pmjdebruijn, 2021-04-12:
@korzyn1: for your own use, you can just add G29 after G28 in your start_gcode and it should be fine.

korzyn1, 2021-04-12:
Yes, I performed that upgrade by myself. I just created a new entry in the configuration file and and got it  perfectly to work. I copied the ender-6 data and added the fastbl variable to it as it is in the ender 3 bltouch version of the prusaslicer 2.4.0 alpha. All is working fine.