prusa3d/PrusaSlicer #6341: Prusa slicer is black

Josebanchero, 2021-04-11:
### Version
_PrusaSlicer-2.0.0_

### Operating system type + version
_macOS Big Sur_

### 3D printer brand / version + firmware version (if known)
_Prusa i3MK3s

### Behavior
* _some parts of the presa slicer turned black, i dont know how it happened or how to fin it, please help!

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
[image]
[image]
[image]

fe60, 2021-04-12:
Do you really run version 2.**0**.0?
The latest version available is 2.**3**.0.
https://github.com/prusa3d/PrusaSlicer/releases/latest

rtyr, 2021-04-12:
Does it happen with 2.3.0?

Josebanchero, 2021-04-12:
i updates to the new versión and it stoped, so thank you!

closed 2021-04-12 by Josebanchero