prusa3d/PrusaSlicer #6342: SIGSEGV (Address boundary error) when slicing particular model

kuon, 2021-04-12:
### Version

Prusa Version 2.3.0. But I tried with prusa master and got same crash.

### Operating system type + version

Archlinux up to date as of April 12

### 3D printer brand / version + firmware version (if known)

Own printer

### Behavior

- Open prusa (attached project).
- Slice.
- Crash occurs after slicing.

Other models work. Only this model crashes the app.

#### Project File (.3MF) where problem occurs

[project.zip](https://github.com/prusa3d/PrusaSlicer/files/6293101/project.zip)

lukasmatena, 2021-04-12:
I cannot reproduce it. Can you please run the application from command line as `./prusa-slicer lloglevel=5` to get verbose logging output? Are you able to provide a backtrace? Thanks.

kuon, 2021-04-12:
I would need to rebuild it with debug symbols on to get a more useful backtrace. But in the meantime here it is.

Actually at a first glance it looks like the bug is not within prusa.

```
#0  0x00007ffff580b494 in pthread_mutex_lock () at /usr/lib/libpthread.so.0
#1  0x00007fffddac58df in  () at /usr/lib/dri/radeonsi_dri.so
#2  0x00007fffddac7ae2 in  () at /usr/lib/dri/radeonsi_dri.so
#3  0x0000555555fcdb00 in  ()
#4  0x0000555555e4e28e in  ()
#5  0x0000555555e4ff06 in  ()
#6  0x0000555555ff9945 in  ()
#7  0x00007ffff6e51dde in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) ()
    at /usr/lib/libwx_baseu-3.0.so.0
#8  0x00007ffff6e5215b in wxEvtHandler::SearchDynamicEventTable(wxEvent&) () at /usr/lib/libwx_baseu-3.0.so.0
#9  0x00007ffff6e521f1 in wxEvtHandler::TryHereOnly(wxEvent&) () at /usr/lib/libwx_baseu-3.0.so.0
#10 0x00007ffff6e522ac in wxEvtHandler::ProcessEventLocally(wxEvent&) () at /usr/lib/libwx_baseu-3.0.so.0
#11 0x00007ffff6e52382 in wxEvtHandler::ProcessEvent(wxEvent&) () at /usr/lib/libwx_baseu-3.0.so.0
#12 0x00005555560c08c3 in  ()
#13 0x00005555560a51ab in  ()
#14 0x00005555560a3c54 in  ()
#15 0x00007ffff6e51dde in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) ()
    at /usr/lib/libwx_baseu-3.0.so.0
#16 0x00007ffff6e5215b in wxEvtHandler::SearchDynamicEventTable(wxEvent&) () at /usr/lib/libwx_baseu-3.0.so.0
#17 0x00007ffff6e521f1 in wxEvtHandler::TryHereOnly(wxEvent&) () at /usr/lib/libwx_baseu-3.0.so.0
#18 0x00007ffff6e522ac in wxEvtHandler::ProcessEventLocally(wxEvent&) () at /usr/lib/libwx_baseu-3.0.so.0
#19 0x00007ffff6e52382 in wxEvtHandler::ProcessEvent(wxEvent&) () at /usr/lib/libwx_baseu-3.0.so.0
#20 0x00005555560c08c3 in  ()
#21 0x00005555560a90a7 in  ()
#22 0x00005555560a9668 in  ()
#23 0x00005555560aa6bc in  ()
#24 0x0000555555d3e7a8 in  ()
#25 0x0000555555d609e1 in  ()
#26 0x00007ffff6e51dde in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) ()
    at /usr/lib/libwx_baseu-3.0.so.0
#27 0x00007ffff6e5215b in wxEvtHandler::SearchDynamicEventTable(wxEvent&) () at /usr/lib/libwx_baseu-3.0.so.0
#28 0x00007ffff6e521f1 in wxEvtHandler::TryHereOnly(wxEvent&) () at /usr/lib/libwx_baseu-3.0.so.0
#29 0x00007ffff6e522ac in wxEvtHandler::ProcessEventLocally(wxEvent&) () at /usr/lib/libwx_baseu-3.0.so.0
#30 0x00007ffff6e52382 in wxEvtHandler::ProcessEvent(wxEvent&) () at /usr/lib/libwx_baseu-3.0.so.0
#31 0x00007ffff6e52e87 in wxEvtHandler::ProcessPendingEvents() () at /usr/lib/libwx_baseu-3.0.so.0
#32 0x00007ffff6cefdb0 in wxAppConsoleBase::ProcessPendingEvents() () at /usr/lib/libwx_baseu-3.0.so.0
#33 0x00007ffff687c58a in wxApp::DoIdle() () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#34 0x00007ffff687c6b4 in  () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#35 0x00007ffff59d3ea0 in g_main_context_dispatch () at /usr/lib/libglib-2.0.so.0
#36 0x00007ffff5a27a49 in  () at /usr/lib/libglib-2.0.so.0
#37 0x00007ffff59d3503 in g_main_loop_run () at /usr/lib/libglib-2.0.so.0
#38 0x00007ffff5ce758f in gtk_main () at /usr/lib/libgtk-3.so.0
#39 0x00007ffff6897b26 in wxGUIEventLoop::DoRun() () at /usr/lib/libwx_gtk3u_core-3.0.so.0
#40 0x00007ffff6d2a37e in wxEventLoopBase::Run() () at /usr/lib/libwx_baseu-3.0.so.0
#41 0x00007ffff6cf0766 in wxAppConsoleBase::MainLoop() () at /usr/lib/libwx_baseu-3.0.so.0
#42 0x00007ffff6d6f199 in wxEntry(int&, wchar_t**) () at /usr/lib/libwx_baseu-3.0.so.0
#43 0x0000555555ce3e9b in  ()
#44 0x00005555557709f8 in  ()
#45 0x000055555575a2d5 in  ()
#46 0x00007ffff565ab25 in __libc_start_main () at /usr/lib/libc.so.6
#47 0x0000555555769b1e in  ()
```

```
[2021-04-12 11:02:32.922620] [0x00007f876bdb5640] [info]    Exporting G-code... Resident memory: 160MB; Shared memory: 90MB; Private memory: 71MB; Peak memory usage: 160MB
[2021-04-12 11:02:32.928338] [0x00007f877cf2d080] [debug]   Loading print object toolpaths in parallel - finalizing results (GLVolumeCollection RAM: 7MB GPU: 0MB Both: 0MB) Resident memory: 167MB; Shared memory: 90MB; Private memory: 77MB; Peak memory usage: 167MB
[2021-04-12 11:02:32.929589] [0x00007f877cf2d080] [debug]   Loading print object toolpaths in parallel - end (GLVolumeCollection RAM: 7MB GPU: 5MB Both: 5MB) Resident memory: 167MB; Shared memory: 90MB; Private memory: 77MB; Peak memory usage: 167MB
[2021-04-12 11:02:32.932415] [0x00007f876bdb5640] [trace]   Exported layer 0 print_z 0.2 Resident memory: 167MB; Shared memory: 90MB; Private memory: 77MB; Peak memory usage: 167MB
[2021-04-12 11:02:32.935396] [0x00007f876bdb5640] [trace]   Exported layer 1 print_z 0.5 Resident memory: 167MB; Shared memory: 90MB; Private memory: 77MB; Peak memory usage: 167MB
[2021-04-12 11:02:32.937690] [0x00007f876bdb5640] [trace]   Exported layer 2 print_z 0.8 Resident memory: 167MB; Shared memory: 90MB; Private memory: 77MB; Peak memory usage: 167MB
[2021-04-12 11:02:32.940008] [0x00007f876bdb5640] [trace]   Exported layer 3 print_z 1.1 Resident memory: 167MB; Shared memory: 90MB; Private memory: 77MB; Peak memory usage: 167MB
[2021-04-12 11:02:32.942345] [0x00007f876bdb5640] [trace]   Exported layer 4 print_z 1.4 Resident memory: 167MB; Shared memory: 90MB; Private memory: 77MB; Peak memory usage: 167MB
[2021-04-12 11:02:32.945148] [0x00007f876bdb5640] [trace]   Exported layer 5 print_z 1.7 Resident memory: 167MB; Shared memory: 90MB; Private memory: 77MB; Peak memory usage: 167MB
[... cut ...]
[2021-04-12 11:02:33.148164] [0x00007f877cf2d080] [trace]   Loaded G-code generated indices buffers, 10MB Resident memory: 178MB; Shared memory: 90MB; Private memory: 88MB; Peak memory usage: 178MB
[2021-04-12 11:02:33.152126] [0x00007f877cf2d080] [trace]   Loaded G-code generated extrusion paths, 2MB Resident memory: 178MB; Shared memory: 90MB; Private memory: 89MB; Peak memory usage: 178MB
[2021-04-12 11:02:33.172790] [0x00007f877cf2d080] [trace]   Refreshed G-code extrusion paths, 2MB Resident memory: 178MB; Shared memory: 90MB; Private memory: 89MB; Peak memory usage: 178MB
fish: Job 1, 'prusa-slicer -loglevel=5' terminated by signal SIGBUS (Misaligned address error)
```

kuon, 2021-04-12:
The part has no infill (wall are thin), and if I set infill percentage to 0%, the crash does not occur. But with 15% infill, it does.

lukasmatena, 2021-04-12:
Thanks. Quite a lot of similar issues have surfaced recently: #5880, #5889, #6202 and some others. The problem leads to the graphics driver. People reported two workarounds: setting `LIBGL_DRI3_DISABLE=true` and downgrading mesa library. You can try if either works for you. It is also possible to turn off hardware acceleration completely, but that will cripple PrusaSlicer performance (`./prusa-slicer --sw-renderer`). Let us know how it went.

kuon, 2021-04-12:
I don't think this is related, but when I run `ctest-v` I get failed tests (with ASAN enabled).

https://dl.goyman.com/pbin/ac94e2f9-7fe8-4b6c-bcde-434446d841f2.txt

And `LIBGL_DRI3_DISABLE=true` works for me.

lukasmatena, 2021-04-15:
Can you guys please try out the new [2.3.1 patch release](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.1-rc) to see if it helps? It has in https://github.com/prusa3d/PrusaSlicer/issues/6345#issuecomment-820503201. Thanks.

bubnikv, 2021-09-25:
No response, closing.

closed 2021-09-25 by bubnikv