prusa3d/PrusaSlicer #6345: Prusa Slicer 2.3.0 crashes with segmentation fault/sigdev

g0r3, 2021-04-12:
### Version
2.3.0

### Operating system type + version
Arch Linux (Rolling Release)

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S

### Behavior
I load a certain model, activate support, and slice it. One or two seconds after slicing the program will crash. The crash doesn't happen if I not activate the printing of support material.

#### Some files for debugging
Log output
[prusa-slicer --loglevel 5.txt](https://github.com/prusa3d/PrusaSlicer/files/6294811/prusa-slicer.--loglevel.5.txt)

The STL file I try to slice (please remove .txt file suffix, i just added it for the upload). I use PETG as material and I activate also support material before slicing.
[screwcap.stl.txt](https://github.com/prusa3d/PrusaSlicer/files/6294814/screwcap.stl.txt)

lukasmatena, 2021-04-12:
Possibly the same as https://github.com/prusa3d/PrusaSlicer/issues/6342

Nachowarrior, 2021-04-12:
I am also on Arch and the last update completely biffed prusa slicer. 2.3.0 (official release)

The slicer is completely useless as it will crash sometimes without making any changes at all, however it will definitely crash with any operation; a scale, a rotate, any infill at all, etc etc.  This is true for all models, or at least the 5 or 6 I have tried to load up, several of which I had just printed 2 days ago with the old version.

I have not had any luck getting anything to output to a log file launching from terminal 'prusa-slicer --loglevel 5 --output prusalog.txt'  
probably user error/incorrect syntax.   I have also tried various workarounds found on several github and other forums to no avail.
best i get out of terminal is "segmentation fault". 

The only thing I can think to do is to uninstall and re-install the old 2.2 from pacman cache. HOWEVER, there are dependencies that i'm too dumb/ don't have time to figure out how to resolve. 

oh, and for some reason the build of the release candidate fails, so I can't even try to use that to see if the fix works.

mistrjirka, 2021-04-14:
I can agree with Nachowarrior. It is completely unusable because every time the file is sliced the software crashes. This is output from the terminal. I also didn´t manage to get anything from 'prusa-slicer --loglevel 5 --output prusalog.txt', but maybe this can help. 
[prusalog2.log](https://github.com/prusa3d/PrusaSlicer/files/6309738/prusalog2.log)

bubnikv, 2021-04-14:
You may guys also try the just released PrusaSlicer 2.3.1-rc. We did some
OpenGL optimization, it may or may not help.

Nachowarrior, 2021-04-14:
> You may guys also try the just released PrusaSlicer 2.3.1-rc. We did some OpenGL optimization, it may or may not help. st 14. 4. 2021 v 11:13 odesílatel mistrjirka ***@***.***> napsal:
> […](#)

The release candidate from the AUR fails to build. 

The only way this currently runs is in software only mode. which is like watching a slideshow even on a modern ryzen processor.

LIBGL_ALWAYS_SOFTWARE=true prusa-slicer

bubnikv, 2021-04-15:
> The release candidate from the AUR fails to build.

You may run our appimage.

g0r3, 2021-04-15:
2.3.1-rc solves the problem for me. Thanks!

mixicz, 2021-04-16:
I can confirm, 2.3.1-rc solves segfault for me as well.

pmz98, 2021-04-16:
Same here 2.3.1-rc solved it on arch linux for me

pjanx, 2021-04-17:
AMD, BTW I use Arch, and I have the crash, too. AUR/master builds, but gives me:
```
ASSERT INFO:
./include/wx/dynarray.h(837): assert "uiIndex < m_nCount" failed in Item().

BACKTRACE:
[1] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[2] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[3] wxEvtHandler::TryHereOnly(wxEvent&)
[4] wxEvtHandler::ProcessEventLocally(wxEvent&)
[5] wxEvtHandler::ProcessEvent(wxEvent&)
[6] wxNotebookBase::SendPageChangedEvent(int, int)
[7] g_closure_invoke
[8] g_signal_emit_valist
[9] g_signal_emit
[10] wxNotebook::InsertPage(unsigned long, wxWindow*, wxString const&, bool, int)
[11] wxEntry(int&, wchar_t**)
[12] __libc_start_main
```
`LIBGL_DRI3_DISABLE=true` doesn't work, `LIBGL_ALWAYS_SOFTWARE=true` works, infill to zero works, 2.3.1rc works:
```
pkgname=prusa-slicer
_pkgver=2.3.1-rc
pkgver=${_pkgver/-/}
pkgrel=1
pkgdesc="G-code generator for 3D printers (Prusa fork of Slic3r)"
arch=(x86_64)
url="https://github.com/prusa3d/PrusaSlicer"
license=(AGPL3)
depends=(boost-libs curl glew intel-tbb nlopt wxgtk3 qhull openvdb cgal)
makedepends=(cmake boost cereal eigen expat gtest libpng systemd)
replaces=(slic3r-prusa3d)
source=(${url}/archive/refs/tags/version_${_pkgver}.tar.gz)
sha256sums=('SKIP')

build() {
  cmake -B build -S PrusaSlicer-version_${_pkgver} \
    -DCMAKE_INSTALL_PREFIX=/usr \
    -DCMAKE_INSTALL_LIBDIR=lib \
    -DSLIC3R_FHS=ON \
    -DSLIC3R_PCH=OFF \
    -DSLIC3R_WX_STABLE=ON \
    -DSLIC3R_GTK=3 \
    -DwxWidgets_CONFIG_EXECUTABLE=/usr/bin/wx-config-gtk3
  make -C build
}

check() {
  cd build
  ctest -v
}

package() {
  make -C build DESTDIR="${pkgdir}" install

  # Desktop files
  mv "${pkgdir}"/usr/share/{PrusaSlicer/,}applications

  # Desktop icons
  for i in 32 128 192 ; do
    mkdir -p "${pkgdir}"/usr/share/icons/hicolor/${i}x${i}/apps/
    ln -s /usr/share/PrusaSlicer/icons/PrusaSlicer_${i}px.png "${pkgdir}"/usr/share/icons/hicolor/${i}x${i}/apps/PrusaSlicer.png
    ln -s /usr/share/PrusaSlicer/icons/PrusaSlicer-gcodeviewer_${i}px.png "${pkgdir}"/usr/share/icons/hicolor/${i}x${i}/apps/PrusaSlicer-gcodeviewer.png
  done
}
```

pjanx, 2021-04-19:
This one still crashes with 2.3.1-rc https://www.thingiverse.com/thing:3105150

mrredcon, 2021-04-19:
I get a crash after slicing [Benchy](https://www.thingiverse.com/thing:763622) with the AppImage and after compiling 2.3.1-rc myself

mrredcon, 2021-04-19:
I just tried out prusa-slicer-git (version prusa-slicer-git 2.3.0.r24.gd06aa6069-2 ) from the AUR and experienced the same exact error as @pjanx .

Nachowarrior, 2021-04-19:
Someone on another thread had success one of two ways.   Dropping your mesa driver back to the stone age ( I think it was 3 or so releases backward) or removing prusa slicer 2.3 entirely and dropping back to 2.2.   The only other way is software only mode, which is painful at best.  TLDR punch this in console:   LIBGL_ALWAYS_SOFTWARE=true prusa-slicer

Nachowarrior, 2021-04-19:
FWIW:  I just received an update to mesa, issue still exists.

bubnikv, 2021-04-19:
I am difficulty to follow the discussion, however our stance to Linux issues is the following:

We only support PrusaSlicer on Linux if running our internal builds or builds compiled against the static dependencies produced by dependency script.

jdrizin, 2021-04-20:
I've been using the AppImages and getting crashes as well. Slicing files leads to crashing unless I disable hardware accel with LIBGL_ALWAYS_SOFTWARE=true on 2.3.0-rc2, 2.3.0, 2.3.1-rc and 2.3.1.

AMD Ryzen 5 3600, AMD 5700XT, kernel 5.11.15, Mesa 21.0.2. Attached is a crash log and project file that causes the crash on 2.3.1 for me

[prusaslicercrashlog.txt](https://github.com/prusa3d/PrusaSlicer/files/6343828/prusaslicercrashlog.txt)
[kkit-lightv4.3mf.txt](https://github.com/prusa3d/PrusaSlicer/files/6343837/kkit-lightv4.3mf.txt)

Nachowarrior, 2021-05-01:
> I've been using the AppImages and getting crashes as well. Slicing files leads to crashing unless I disable hardware accel with LIBGL_ALWAYS_SOFTWARE=true on 2.3.0-rc2, 2.3.0, 2.3.1-rc and 2.3.1. [...]

yeah, it's been over 3 weeks since this bug started. I contacted prusa directly and they said they couldn't help me...   

I'm beginning to rethink my choice of purchase entirely if we're going to be left hanging for so long.   The official repositories have not been updated so there's nothing to do but wait unless you want to blow all your precious time fiddling with some stupid workaround.   Super disappointed in how this got swept under the rug for so long.

TLDR: I really don't think they care to support linux users, tbh.

mrredcon, 2021-05-06:
Updating to prusa-slicer-2.3.1-1 and mesa 21.0.3-3 on Arch I am no longer experiencing a crash when slicing Benchy.  Problem solved for me!

jkdrz, 2021-05-06:
Mesa 21.0.3-3 on Arch also solves the crashes for me.

Legodev, 2021-05-23:
Used the AppImage of 2.3.1 it worked for a long time, but now i encountered the crash again.
I wanted to slice the normal EVA fanduct:
https://main.eva-3d.page/addons/fan-ducts/stls/Duct%20v2.1%20Straight.stl

If i lay it flat, such as the default orientation of the stl and then slice it, it works fine.
If i rotate it upright, so that it becomes printable, prusa slicer crashes.

It works when i use the `LIBGL_ALWAYS_SOFTWARE=true`