prusa3d/PrusaSlicer #6346: Print Settings \ Advanced \ Seam Position error

DrJuJu, 2021-04-12:
### Version

Version 2.3.0+win64

### Operating system type + version

Edition	Windows 10 Pro for Workstations
Version	20H2
Installed on	‎11/‎06/‎2020
OS build	19042.870
Experience	Windows Feature Experience Pack 120.2212.551.0

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior

Trying to select a RANDOM seam position always results in a model with a vertical seam !!  no matter what model is being sliced or ant pre-configured "Print Settings" settings is selected, & the seam painting tool is useless as it's impossible to work out where the seams end up !!

lukasmatena, 2021-04-12:
Please post a 3MF.

DrJuJu, 2021-04-12:
I'm unable to, as I never use 3MF files ...

Hooch180, 2021-04-12:
@DrJuJu - In Prusas Slicer just go to File->Save Project and upload the 3mf project here.

DrJuJu, 2021-04-12:
I also suggest you get Github to allow direct uploading of the 3MF files you require

[Samyang_Solar_Filter_Body_v2.zip](https://github.com/prusa3d/PrusaSlicer/files/6296195/Samyang_Solar_Filter_Body_v2.zip)

closed 2026-09-29 by prusaslicer-bot[bot]