prusa3d/PrusaSlicer #6348: Cannot build dependencies: boost is not available anymore

krisrok, 2021-04-12:
Was hosted at bintray.org, but it's lifetime ends pretty soon: https://jfrog.com/blog/into-the-sunset-bintray-jcenter-gocenter-and-chartcenter/

lukasmatena, 2021-04-13:
This should be fine after #6349 was merged. Thanks, closing.

closed 2021-04-13 by lukasmatena

erickwill, 2021-04-13:
> Was hosted at bintray.org, but it's lifetime ends pretty soon: https://jfrog.com/blog/into-the-sunset-bintray-jcenter-gocenter-and-chartcenter/

Did you get to build it? Are you building for Linux?

krisrok, 2021-04-13:
No, for windows and yes, it worked. It downloads the same file (same hash) after all. 

@lukasmatena replaced the url for other platforms: https://github.com/prusa3d/PrusaSlicer/commit/279113900209ba624c71020fde22ea9ac089e04e

erickwill, 2021-04-13:
> No, for windows and yes, it worked. It downloads the same file (same hash) after all.
> 
> @lukasmatena replaced the url for other platforms: [2791139](https://github.com/prusa3d/PrusaSlicer/commit/279113900209ba624c71020fde22ea9ac089e04e)

Nice! Can you share your executable?