prusa3d/PrusaSlicer #6351: Application Hangs - Mac M1

stefanpieter, 2021-04-12:
### Version
Prusa Slicer 2.30+

### Operating system type + version
Mac Mini 16GB RAM running 11.2.2 (20D80)

### Behavior
I import an STL file from Fusion 360 using the "3D Print" feature. All goes well until you hit slice. The slicer then hangs and is unreachable. I left it for a day to compute without luck. One has to kill the program. This happens every time on multiple files.

https://user-images.githubusercontent.com/65555837/114446836-970ee680-9bc9-11eb-9e6d-ba847a006be3.mov

fe60, 2021-04-13:
Can you share the stl-file and/or the project file (File=>Export=>Export plate as AMF...)?
Without those files it will be hard for the developers to reproduce the problem.
You will have to zip the files in order to attach them to this issue.

stefanpieter, 2021-04-13:
Hello @fe60 

As requested.
This is happening with many different models and files. The attached is just one example.

[testing_rig_left_rear_C14.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/6301682/testing_rig_left_rear_C14.3mf.zip)

bubnikv, 2021-04-14:
Should be fixed with the upcoming PrusaSlicer 2.3.1-rc

closed 2021-04-14 by bubnikv