prusa3d/PrusaSlicer #6353: Hanging support

burchesoka, 2021-04-13:
### Version
_Version of PrusaSlicer used goes here_ 2.3.0

_Use `About->About PrusaSlicer` for release versions_Version 2.3.0+win64

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_Windows 10 Pro
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_Hi, very often I see hanging support, such as in image.
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_[image]

[image]

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2021-04-14:
Unfortunately, it is not possible to upload files via e-mail attachements. You have to upload it using github web interface (archive it to zip first) .

burchesoka, 2021-04-15:
[111111.zip](https://github.com/prusa3d/PrusaSlicer/files/6315615/111111.zip)

rtyr, 2021-05-13:
You probably sent different project. It doesn't correspond with your screenshot (also layer/height doesn't match). I don't see the problem you were refering to.

bubnikv, 2021-10-05:
No response, closing.
The supports were improved in PrusaSlicer 2.4.0-alpha3.

closed 2021-10-05 by bubnikv