prusa3d/PrusaSlicer #6354: Build for Linux

erickwill, 2021-04-13:
### Version
version_2.3.1-beta

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Arch Linux - 5.11.13-arch2-1

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior

While trying to build PrusaSlicer for Linux I found the following:
[https://pastebin.com/dcAr07Yz](url)

Do 'm missing something out?
[CMakeError.log](https://github.com/prusa3d/PrusaSlicer/files/6305892/CMakeError.log)
[CMakeOutput.log](https://github.com/prusa3d/PrusaSlicer/files/6305893/CMakeOutput.log)

lukasmatena, 2021-04-14:
I'd start by reading https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md.

erickwill, 2021-04-14:
Reinstalling all dependencies did the trick. It's building nicely now. Thanks.

closed 2021-04-14 by lukasmatena