prusa3d/PrusaSlicer #6355: TBB cmake file needs updating to latest, fix included

theofficialgman, 2021-04-14:
Current findTBB.cmake used it outdated and does not match the head. Certain TBB packages don't include the debug headers causing a mismatch in the TBB chosen.

Issue discussed here: https://github.com/ceres-solver/ceres-solver/issues/612

It was fixed in the upstream TBB cmake last year (commit found here: https://ceres-solver-review.googlesource.com/changes/ceres-solver~17720/revisions/2/files/cmake%2FFindTBB.cmake/download)

Replacing with this new cmake file has no issues.

loqs, 2021-05-25:
@theofficialgman did https://github.com/loqs/PrusaSlicer/commit/b269a7f541fe62770e68660834e4a36d37db1ce4 resolve the issue for you?  Did the rest of the series help with building with TBB2021?

theofficialgman, 2021-05-26:
no, those changes in that branch do not solve the issue. The cmake file which I originally referenced works fine

loqs, 2021-05-26:
So it needs two fixes then as the cmake file you linked to is TBB2021 incompatible.

theofficialgman, 2021-05-26:
yes I only ever tested tbb2017 (bionic's package version) and tbb2020(hirsute's package)
I was not aware the 2021 changes. so yes, both of those commit changes would need to be included

loqs, 2021-05-26:
I think the change of license and copyright would need to be looked at.

theofficialgman, 2021-05-26:
ah well, the PrusaSlicer developers are welcome to rewrite this cmake file as they please
I was mostly making this issue to bring attention to what I ran into and hoping for some response from the devs.

loqs, 2021-05-26:
https://github.com/loqs/PrusaSlicer/commit/cbeb1d48fafbc8b655cdef65c2cc1795ea36ae60 applies the current VTK version of the file in the same manner ceres-solver did then I applied their changes as two separate commits.

theofficialgman, 2021-05-26:
I assume that change will work now since master+manually replacing the cmake file works fine. 
Were the other edits you made in the TBB2021 branch necessary to get TBB2021 functional (talking about the first 5commits)?

loqs, 2021-05-26:
Yes TBB2021 dropped a lot of functions that were previously deprecated. [1] provides suggested replacements for most of the functions.

[1] https://software.intel.com/content/www/us/en/develop/articles/tbb-revamp.html

bnavigator, 2021-11-22:
Any progress on this? We are also hitting this after openSUSE Tumbleweed has updated to TBB 2021

theofficialgman, 2021-11-22:
@bnavigator I made a PR of @loqs changes and noted a couple of things that still need changing inorder for it to work (at least for me): https://github.com/prusa3d/PrusaSlicer/pull/7149

I don't have commit access to his repo so I can't make them and they haven't been active.
I might have to cherry pick his commits into my own fork and PR from there inorder to get this moving

tamasmeszaros, 2021-12-14:
Please see #7470 and reopen if you have any questions.

theofficialgman, 2021-12-14:
@tamasmeszaros testing on the lastest changes. I get errors when using cmake related to TBB

```
cmake .. -DSLIC3R_WX_STABLE=1 -DSLIC3R_GTK=3 -DCMAKE_C_COMPILER=gcc-11 -DCMAKE_CXX_COMPILER=g++-11

-- The C compiler identification is GNU 11.1.0

-- The CXX compiler identification is GNU 11.1.0

-- Detecting C compiler ABI info

-- Detecting C compiler ABI info - done

-- Check for working C compiler: /usr/bin/gcc-11 - skipped

-- Detecting C compile features

-- Detecting C compile features - done

-- Detecting CXX compiler ABI info

-- Detecting CXX compiler ABI info - done

-- Check for working CXX compiler: /usr/bin/g++-11 - skipped

-- Detecting CXX compile features

-- Detecting CXX compile features - done

-- No build type selected, default to Release

-- SLIC3R_ASAN: OFF

-- SLIC3R_BUILD_SANDBOXES: OFF

-- SLIC3R_BUILD_TESTS: ON

-- SLIC3R_DESKTOP_INTEGRATION: ON

-- SLIC3R_FHS: OFF

-- SLIC3R_GTK: 3

-- SLIC3R_GUI: ON

-- SLIC3R_MSVC_COMPILE_PARALLEL: ON

-- SLIC3R_MSVC_PDB: ON

-- SLIC3R_PCH: ON

-- SLIC3R_PERL_XS: OFF

-- SLIC3R_PROFILE: OFF

-- SLIC3R_STATIC: OFF

-- SLIC3R_WX_STABLE: 1

-- SLIC3R_STATIC: OFF

-- CMAKE_PREFIX_PATH: (default)

-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 

-- Looking for pthread.h

-- Looking for pthread.h - found

-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
[... 162 lines cut ...]

-- Boost include dirs: /usr/include

-- Boost libraries:    

-- Using gcc version 4 or later. Adding -frounding-math

-- Adding precompiled header pchheader.hpp to target libslic3r.

-- Found wxWidgets: -L/usr/lib/aarch64-linux-gnu;-pthread;;;-lwx_baseu-3.0;-lwx_gtk3u_core-3.0;-lwx_gtk3u_adv-3.0;-lwx_gtk3u_html-3.0;-lwx_gtk3u_gl-3.0 (found suitable version "3.0.4", minimum required is "3.0") 

-- wx-config path: /usr/bin/wx-config

-- wx libs: -L/usr/lib/aarch64-linux-gnu;-pthread;;;-lwx_baseu-3.0;-lwx_gtk3u_core-3.0;-lwx_gtk3u_adv-3.0;-lwx_gtk3u_html-3.0;-lwx_gtk3u_gl-3.0;/usr/lib/aarch64-linux-gnu/libpng.so;/usr/lib/aarch64-linux-gnu/libz.so;/usr/lib/aarch64-linux-gnu/libtiff.so;/usr/lib/aarch64-linux-gnu/libjpeg.so;/usr/lib/aarch64-linux-gnu/libexpat.so;/usr/lib/aarch64-linux-gnu/librt.so

-- Adding precompiled header pchheader.hpp to target libslic3r_gui.

-- Found GTK3: /usr/local/include/gtk-3.0;/usr/include/at-spi2-atk/2.0;/usr/include/at-spi-2.0;/usr/include/dbus-1.0;/usr/lib/aarch64-linux-gnu/dbus-1.0/include;/usr/include/gio-unix-2.0/;/usr/include/cairo;/usr/include/pango-1.0;/usr/include/harfbuzz;/usr/include/pango-1.0;/usr/include/atk-1.0;/usr/include/cairo;/usr/include/pixman-1;/usr/include/freetype2;/usr/include/libpng16;/usr/include/freetype2;/usr/include/libpng16;/usr/include/gdk-pixbuf-2.0;/usr/include/libpng16;/usr/include/glib-2.0;/usr/lib/aarch64-linux-gnu/glib-2.0/include  

-- Configuring incomplete, errors occurred!

See also "/home/garrett/PrusaSlicer/build/CMakeFiles/CMakeOutput.log".

See also "/home/garrett/PrusaSlicer/build/CMakeFiles/CMakeError.log".
```

theofficialgman, 2021-12-14:
@tamasmeszaros found the bug. for some reason a "FindTBB" is executed in the main project CMakelists.txt in addition to your module which is already added

https://github.com/prusa3d/PrusaSlicer/blob/caff954aa78c4e0d5a170ab18802af3c2f66ebed/CMakeLists.txt#L384
that needs to be disabled

tamasmeszaros, 2021-12-16:
@theofficialgman Can you try the above commit please?

theofficialgman, 2021-12-16:
@tamasmeszarosI checked out the tm_tbb_find_fix branch and that does successfully complete the cmake

I still don't understand why you don't remove it from the main CMakeLists.txt though?

theofficialgman, 2021-12-16:
@tamasmeszaros (made a typo with your username, response in the above message)

tamasmeszaros, 2021-12-17:
Ok @theofficialgman I'm glad it works. find_package call in the main CMakeLists.txt is required as TBB is a dependency of PS. It only happens that FindOpenVDB also needs TBB and calls FindTBB first, but that does not mean we should omit our statement in our cmake. (BTW OpenVDB cmake is a mess.)

That being said, we will not have the resources in the future to solve issues like this, unless you are a package maintainer. It is an impossible task to ensure a successful build on every version of every distro, not to mention the issues arising from using different versions of libraries than we use to test the final executable. We have community maintained snap and flatpak packages which should suffice every distro that has no native package of PrusaSlicer. We can commit our time to support the maintainers of these two packages and a handful of maintainers of majors distros for native packages.

If you want to build from the latest master, please use the dependency build scripts, because we can at least work on those to always link and compile successfully on supported platforms.

closed 2021-12-17 by tamasmeszaros

theofficialgman, 2021-12-17:
@tamasmeszaros I appreciate the help. Will this change be pulled into master though (I don't see it as of right now)?

I'm currently using an arm64 only system so can't run any of the official packages (besides the outdated prusaslicer version provided by ubuntu). I'll look into using the dependency build scripts in the future or submitting PRs for fixes in source if they arise again.

tamasmeszaros, 2021-12-20:
Hello, so given that it works for you, I will merge it into master after our major release.

theofficialgman, 2021-12-27:
@tamasmeszaros looks like the 2.4.0 major release has been made 🥳
I'd appreciate if this was merged

theofficialgman, 2022-01-15:
@tamasmeszaros could you please merge the branch? https://github.com/prusa3d/PrusaSlicer/tree/tm_tbb_find_fix

commit 914f170a (2021-12-14): Use TBB config scripts to find the library if using dynamic libraries.

fixes #7470
fixes #6355 ?
commit eddcebf9 (2022-02-03): Fix repeated calls to FindTBB in module mode.

fixes #6355