prusa3d/PrusaSlicer #6356: Place on face creates slicing issues with 0.8 and 0.6 nozzles depending on layer height selected

reven, 2021-04-14:
### Version
Version 2.3.0+win64
Build: PrusaSlicer-2.3.0+win64-202101111315

### Operating system type + version
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19042), 64-bit edition

### 3D printer brand / version + firmware version (if known)
Prusa MK3s (unmodified)
FIRMWARE_NAME:Prusa-Firmware 3.9.3

### Behavior
* An stl mesh (file _VER in attached project) verified in Blender and Meshmixer as being manifold, having no duplicated vertices and correct normals, fails to slice correctly and creates artifacts when using the "Place on face" function.
* The same model exported horizontaly (file _HOR) slices correctly with any settings combination (nozzle or layer height)
* **Expected**: there should be no difference in slicing between the two models.
* Vertical and placed on face image:
[image]
* Horizontal image:
* [image]

#### Project File (.3MF) where problem occurs
[Place on face error.zip](https://github.com/prusa3d/PrusaSlicer/files/6308933/Place.on.face.error.zip)

rtyr, 2021-04-14:
https://github.com/prusa3d/PrusaSlicer/wiki/Frequently-Asked-Questions-(FAQ)

A perfectly flat surface may not stay perfectly flat after being rotated in a 3D modelling software. If the slicer happens to slice across such a no more perfectly flat surface, weird one layer thick islands may show up. 

See #1448 and #2434 for further details.

Closing.

closed 2021-04-14 by rtyr

reven, 2021-04-14:
Thanks, I think I understand much better whats happening now, I still feel this "rounding error" is being introduced by the place on face function in this particular case.