prusa3d/PrusaSlicer #6358: Layer shifting problem with CR10

JamieClaye, 2021-04-14:
Version 2.3.0+win64
- Prusa MK3+ MMU2
- Creality CR10

### Behavior
I get random layer shifting with a model on the CR10 but not on the Prusa.  However if I slice the same model using Cura for the CR10 it prints without layer shifting.  I know the CR10 isn't your printer (of course) but I really like Slic3rPE and would like to resolve what is going on here.  

I've included the file below. 

[Hatch.zip](https://github.com/prusa3d/PrusaSlicer/files/6312513/Hatch.zip)

rtyr, 2021-04-14:
Have you tried built-in CR10 profile?

pmjdebruijn, 2021-04-15:
Presumably we're not talking about a CR-10 but a CR-10 V3...

How long do both prints take? I'm guessing the Cura print probably takes noticeably longer?

The profile embedded in the 3mf, uses fairly high accelerations and speeds (at least for creality printer):
; travel_speed = 200
; machine_max_acceleration_x = 9000,1000
; machine_max_acceleration_y = 9000,1000
Essentially it seems the third party printer profile depends on the firmware to impose limits, which it may or may not do.

As @rtyr suggested, you should try the CR-10 V3 profile that was shipped with PrusaSlicer 2.3.0 (though I highly recommend 2.3.1-rc as other improvements have been made).

Which brings me to retract_length, which is currently set to 1mm for the CR-10 V3, would you mind printing the retraction test below, at retract_length 1.0, 1.5 and 2.0 in regular pla, and reporting what the lowest length is that's free of stringing?
https://www.thingiverse.com/thing:909901

JamieClaye, 2021-04-16:
Thanks - I thought I did but maybe I grabbed one from a previous built.   I have both CR10 V3 and Ender 3 (converted to direct drive)

Now I have to double check.  The profile I've been using has been working well - just hit a problem with this model.  And I'll run the retract test.    

I'll let you know after I give it a review.

rtyr, 2021-05-13:
Can we close this issue?

rtyr, 2021-06-15:
No response. Closing.

closed 2021-06-15 by rtyr