prusa3d/PrusaSlicer #6360: [Feature Request] Wizard for newbies to learn how to work with the program

Tillema, 2021-04-14:
### Version
Latest version, not relevant.

### Description
Today, I introduced our MK3S+ to my colleagues. While setting up PrusaSlicer, everyone was ready, and I had to go for a pee. After coming back, they were still staring at the same screen. At that point I realized, I had to guide them through the program.
Especially for newbies, the program might be difficult to understand on the first go. Some kind of guide with tips, tricks, etc. will give them a boost. Currently, the program is not yet at the level of a CAD program, but for newbies, it might be overwhelming when they see the thousands of options you have, even in simple mode.
It's difficult to explain how I have it in my mind, so, at this my video editing skills come into practice:

https://user-images.githubusercontent.com/43646872/114762613-3bc42c00-9d62-11eb-9e3b-6f6946865336.mp4

_Is this a new feature request?_
I'm not good at Googling, but yes, this one didn't occur in the search results.

bubnikv, 2021-04-15:
Prusa Knowledge Base (prusa3d.com)
<https://help.prusa3d.com/en/category/prusaslicer_204>

bubnikv, 2021-04-15:
I just checked, we are missing a link to open the web page I have
referenced. And I agree with you that a quick "how to start" video would be
helpful.

bubnikv, 2021-04-15:
We have a "first print" manual page
https://help.prusa3d.com/en/article/first-print-with-prusaslicer_1753

mikolaszuza, 2021-04-15:
@Tillema Good feature request, thanks.
If you want, I can give you access to "Prusa Academy" - a step-by-step guide for (not only) PrusaSlicer with a quiz after each chapter. It's currently in a beta stage.
I'll just need your email (feel free to send it to mikolas.zuza@prusa3d.com if you don't want to publically share it here).

Though a separate video that you suggested would also be helpful, I'll see if I find the time to make one in the near future.

pmjdebruijn, 2021-04-15:
It does beg the question is whether it should be "in" PrusaSlicer, given YouTube is a thing...

After a user first installs PrusaSlicer, he's greeted with the "Welcome to the PrusaSlicer Configuration Wizard", with very little on this initial page. It might make a lot of sense, to overtly advertise a curated YouTube playlist right there. 

The problem with introductory "sessions" as implied by the original poster, is that they usually tend to be very obtrusive, and infuriating to anybody who's already familiar with PrusaSlicer, or any other slicer at all really.

Tillema, 2021-04-15:
I didn't think about a video. An interactive guide would be more helpful, like "click here", "do this", "scale to 50%", etc.
Of course, it should be optional, so, users that are already familiar with other slicers or earlier versions of PS can skip the introduction.
What I think is a good way to introduce is a last page in the initial configuration wizard were is stated "All is set, happy printing", with an option to check a box that says something like "Guide me through the program".

bubnikv, 2021-04-19:
We plan to add some kind of "did you know" functionality one day, possibly
through notifications on application start-up.

neophyl, 2021-04-20:
Just had a similar comment in #6386 .  The knowledge base for PS is not well advertised or obvious at all.  Prusa has spent all this effort creating it and doesn't really advertise that fact.  The Help could really do with a Direct link to the Knowledge base too.

mikolaszuza, 2021-04-20:
You can actually click the individual parameters and the correct documentation will be opened.
[image]

This made some users really unhappy though #5196, so there is now a confirm dialog.

closed 2026-09-29 by prusaslicer-bot[bot]