prusa3d/PrusaSlicer #6361: Incorrect temperature exported for MK3S+MMU2S

jtebert, 2021-04-14:
### Version
2.3.0+linux+x64-GTK3

### Operating system type + version
Pop!_OS 20.10

### 3D printer brand / version + firmware version (if known)
i3 MK3S + MMU2S

### Behavior

When slicing the model with the default MK3S+MMU2S printer profile, it exports Gcode with the incorrect temperature when using extruder 3 only.

In the attached 3MF file, that extruder is set to the Generic PETG filament, but the resulting Gcode has temperature set to the default PLA temperatures (215/60, then 205/60). 

This *only* occurs with extruder 3, but it occurs with MK3S+MMU2S printer profiles for both nozzle sizes (0.4 and 0.6mm).

However, if I change the filament for that extruder to something else (even a saved copy of the Generic PETG), the gcode will export with the correct temperature. I'd guess that there's an issue with default values somewhere, which gets overridden when the filament is changed.

#### Project File (.3MF) where problem occurs

[temp-test.zip](https://github.com/prusa3d/PrusaSlicer/files/6314044/temp-test.zip)

rtyr, 2021-04-15:
Extruder 3 is set to Prusament PLA, so exported temperature is OK.

[image]

Only issue I can see here is the material type in the exported filename. It is known issue https://github.com/prusa3d/PrusaSlicer/issues/5300.

jtebert, 2021-04-15:
Interesting. Because when I open that same file, it shows the filament for that extruder as Generic PETG.

[image]

This is the case with the 3MF file I have locally, as well as re-downloading the one I posted here.

tol2cj, 2021-04-21:
I have similar problem today, with extruder 4. Extruder 4 material is set to Prusament PETG, Object is set to extruder 4, but tempreratures in gcode are PLA. Dont know how I got to this state, but restart helped.

rtyr, 2021-09-08:
Could you try to open that 3MF file in https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.0-alpha1?

rtyr, 2021-09-28:
No response. I was not able to reproduce it with current alpha. Closing.

closed 2021-09-28 by rtyr

jtebert, 2021-11-11:
I just saw this and tried opening the file with the 2.4.0-beta1 release. This 3MF file is showing up as Prusament PLA, which seems correct. I tried reproducing the bug from scratch with this slicer version as well, and the temperatures do set correctly. So yes, it appears this has been resolved with some change in the new version.