prusa3d/PrusaSlicer #6362: Blobs on print after manual color change

MTBehnke, 2021-04-15:
### Version
2.3.0+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa MK2S (stock), Firmware 3.2.3

### Behavior
* When making a manual (single-extruder) color change using PrusaSlicer, after changing filament the printer deposits a large blob at the start of printing the new layer.
* Example photo included below showing the blob around the letter 'R' (white with black letters). When I comment out the gcode line shown below I don't get the blob (black with white letters)
* Tried many different settings in PrusaSlicer but unable to find a way to prevent this issue.
* If this gcode command is vital, consider enabling single extruder printers to use a wipe tower.
```
;COLOR_CHANGE,T0
M600
; printing object Blob Test.stl id:0 copy 0
G1 X82.563 Y68.958
G1 Z2.150
;G1 E1.40000 F3000.000  <--- Commented out for black with white letters example.
M204 S600
```

The small blob and string from the upper left corner (starting position after color change) to where it started printing the letters is a minor issue and easily removed. Also, the gap in the letter 'M' shows up as a gap in PrusaSlicer as a result of the small letters (2.5mm letters), not a printer issue.

[image]

_Is this a new feature request?_No

bubnikv, 2021-04-15:
Fixed with 48a93e40fb2499832ea4dd61185c58912f6ef18a

    After issuing the color change custom G-code, which is most likely
    just M600, reset the internal retract counter, so that a retract
    will happen after the firmware returns from M600 to the initial position.

```
;WIPE_END
G1 E-0.04800 F4200.000
G1 Z7.900 F9000.000
;AFTER_LAYER_CHANGE
;7.85
;COLOR_CHANGE,T0
M600
; printing object Shape-Box id:0 copy 0
**; The following line is newly issued after M600 returns back to the position before M600 with fully pressurized (unretracted) extruder: ***
**G1 E-3.20000 F4200.000**
G1 X81.643 Y98.357 F9000.000
G1 Z7.850
G1 E3.20000 F2400.000
M204 S800
```

closed 2021-04-15 by bubnikv

bubnikv, 2021-07-02:
Fixed with PrusaSlicer 2.3.2-beta.