prusa3d/PrusaSlicer #6364: MacOS / Minimize window / Shortcut CMD + M

Sidirius, 2021-04-15:
### Version
2.3.0

### Operating system type + version
MacOS 11.2.3

### 3D printer brand / version + firmware version (if known)
Doesn't matter

### Behavior
* CMD + M to minimize the window does not work
* _Steps needed to reproduce the problem_
  * Open PrusaSlicer
  * ress CMD + M on the keyboard
* _Expected Results_
  * PrusaSlicer should minimize to the dock
* _Actual Results_
  * Nothing happens

_Is this a new feature request?_
* maybe

kr15uk, 2021-04-29:
Also on "minimised" topic.
If you have minimised app window and you click on the dock icon it should un-minimise the window, doesn't happen with PS 2.3.1 arm.

mcwienczek, 2021-07-31:
I am having similar problem that after minimizing the app, I cannot un-minimize it anymore. PrusaSlicer 2.3.3 on MacBook Pro 16'' MacOs Bigsur 11.5.1

blue68camaro-git, 2022-02-07:
This is still an issue with Version 2.4.1-beta+arm64

FidelCapo, 2022-02-22:
This issue should be fixed in version 2.5.0

closed 2022-02-22 by FidelCapo