prusa3d/PrusaSlicer #6366: Adding Height Range Modifier and speed change affects whole model

lunatic1972, 2021-04-15:
Version 2.4.0-alpha0

I was attempting to print a part off and wanted layers up until 2.0mm at a slower speed and after that go back to its default speed however it seemed to change the speed for the rest of the layers also.
1. I added the model
2. Clicked on "Change object settings->Height Range modifier" with default settings of 0 and 2.0 for height
3. Clicked on "Range 0.00 to 2.00" change object settings->Add Settings->Speed
4. Changed accordingly and sliced the model.
As you can see, it affected the whole model rather than just layers 0.0mm to 2.0mm.
Is this the wrong way to go about it or is this an actual fault in the program??

Thanks

[image]
[Fiber Optic Holder FU-2540_Radwell_part2.zip](https://github.com/prusa3d/PrusaSlicer/files/6320278/Fiber.Optic.Holder.FU-2540_Radwell_part2.zip)

rtyr, 2021-04-15:
The described behavior is likely caused by your cooling settings (Filament settings -> Cooling). You can simply test it by disabling "enable auto cooling" option.

Provided 3MF project doesn't contain any speed modifier (only height range without any additional settings).

[image]

lunatic1972, 2021-04-16:
This one looks like it was my bad, thanks for clarifying that. It makes sense :). You can close this one out.

closed 2021-04-16 by rtyr

lunatic1972, 2021-04-16:
I tried a different model this time and disabled auto cooling as you suggested but the results were the same. Perhaps you could take another look please.
Thanks
[The Warbird - Dragon_fixed.zip](https://github.com/prusa3d/PrusaSlicer/files/6327180/The.Warbird.-.Dragon_fixed.zip)