prusa3d/PrusaSlicer #6367: PrusaSlicer 2.4.0-alpha0 fails to build with gcc 11

ajacocks, 2021-04-15:
### Version
2.4.0-alpha0

### Operating system type + version
Fedora release 34 (Thirty Four)

### 3D printer brand / version + firmware version (if known)
Prusa i3 Mk2.5S

### Behavior
The source code fails to build, on Fedora 34, with the following errors:

/home/ajacocks/rpmbuild/BUILD/PrusaSlicer-version_2.4.0-alpha0/src/libslic3r/Format/3mf.cpp:2390:102: error: expected primary-expression before '&' token
 2390 |     bool _3MF_Exporter::_add_mesh_to_object_stream(mz_zip_writer_staged_context &context, ModelObject& object, VolumeToOffsetsMap& volumes_offsets)
      |                                                                                                      ^
/home/ajacocks/rpmbuild/BUILD/PrusaSlicer-version_2.4.0-alpha0/src/libslic3r/Format/3mf.cpp:2390:104: error: 'object' was not declared in this scope; did you mean 'ObjectID'?
 2390 |     bool _3MF_Exporter::_add_mesh_to_object_stream(mz_zip_writer_staged_context &context, ModelObject& object, VolumeToOffsetsMap& volumes_offsets)
      |                                                                                                        ^~~~~~
      |                                                                                                        ObjectID
/home/ajacocks/rpmbuild/BUILD/PrusaSlicer-version_2.4.0-alpha0/src/libslic3r/Format/3mf.cpp:2390:130: error: expected primary-expression before '&' token
 2390 |     bool _3MF_Exporter::_add_mesh_to_object_stream(mz_zip_writer_staged_context &context, ModelObject& object, VolumeToOffsetsMap& volumes_offsets)
      |                                                                                                                                  ^
/home/ajacocks/rpmbuild/BUILD/PrusaSlicer-version_2.4.0-alpha0/src/libslic3r/Format/3mf.cpp:2390:132: error: 'volumes_offsets' was not declared in this scope; did you mean 'VolumeToOffsetsMap'?
 2390 |     bool _3MF_Exporter::_add_mesh_to_object_stream(mz_zip_writer_staged_context &context, ModelObject& object, VolumeToOffsetsMap& volumes_offsets)
      |                                                                                                                                    ^~~~~~~~~~~~~~~
      |                                                                                                                                    VolumeToOffsetsMap
/home/ajacocks/rpmbuild/BUILD/PrusaSlicer-version_2.4.0-alpha0/src/libslic3r/Format/3mf.cpp:2390:147: error: expression list treated as compound expression in initializer [-fpermissive]
 2390 |     bool _3MF_Exporter::_add_mesh_to_object_stream(mz_zip_writer_staged_context &context, ModelObject& object, VolumeToOffsetsMap& volumes_offsets)
      |                                                                                                                                                   ^
gmake[2]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/build.make:410: src/libslic3r/CMakeFiles/libslic3r.dir/Format/3mf.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[2]: Leaving directory '/home/ajacocks/rpmbuild/BUILD/PrusaSlicer-version_2.4.0-alpha0/x86_64-redhat-linux-gnu'
gmake[1]: *** [CMakeFiles/Makefile2:1172: src/libslic3r/CMakeFiles/libslic3r.dir/all] Error 2
gmake[1]: Leaving directory '/home/ajacocks/rpmbuild/BUILD/PrusaSlicer-version_2.4.0-alpha0/x86_64-redhat-linux-gnu'
gmake: *** [Makefile:163: all] Error 2
error: Bad exit status from /var/tmp/rpm-tmp.zAC7iV (%build)

I suspect that this is due to changed semantics for GCC 11+

#### Project File (.3MF) where problem occurs
We don't get that far.

hejllukas, 2021-05-01:
Do you still have this issue with compilation? I tried to compile PrusaSlicer with GCC 11.1, even on Fedora 34, and I didn't have this issue. There was one missing include, which I added in 62592cab48cfb6a20d84041b1992aecc6a2b659c.
When this issue was created, GCC 11 had not yet released a stable version (even a release candidate wasn't released). So it's possible that the current stable release no longer has this issue.

rtyr, 2021-07-23:
No response. Closing.

closed 2021-07-23 by rtyr