prusa3d/PrusaSlicer #6370: Transparency/hide modifiers

nsarzyns, 2021-04-16:
### Version
2.3.0

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Ender 3 Pro

### Behavior
When placing a modifier for a part, a nice added feature would be the ability to hide or adjust the transparency of that placed modifier. I know you can hide the part as a whole but not the added modifiers. This would help with alignment of the modifier, in the cases I've ran into where this would be useful.

_Is this a new feature request?_
Yes

#### Project File (.3MF) where problem occurs
N/A

nsarzyns, 2021-04-16:
I should state, the transparency would apply to when you are moving the modifier around. I do know that modifiers become transparent when the main part is not the focused part.

rtyr, 2021-04-16:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3678 and https://github.com/prusa3d/PrusaSlicer/issues/4438. Closing.

closed 2021-04-16 by rtyr