prusa3d/PrusaSlicer #6376: nozzle raises up too high when starting to print

AKO22, 2021-04-17:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

AKO22, 2021-04-17:
Uploaded a Thingiverse stl, sliced it in PrusaSlicer, went to run on printer (LotmaxxSC10Shark) , nozzle will raise up too high and start to print. I have auto leveled printer bed, corrected Z height to paper thickness.
Not working, is there a setting that  Pursa Slicer raises the height? I have the same problem in Cura. When I use an older file, it will print good, file was sliced in a Lotmaxx version of Cura. I thing its somewhere in the gcode thats generated... Have deleted both slicers and the registry using "CC cleaner to load new slicers on , still have same problem..

rtyr, 2021-04-18:
Your printer profile is most likely incorrectly configured, the main issue is probably in start g-code section.

AKO22, 2021-04-18:
Morning rtyr, this  is what I found out.I attached a comparison of the two gcodes that were sliced by an old version of Lotmaxx Cura and the new 4.8.0 Cura.
In using Lotmaxx's new version 4.8.0 Cura & Ultimaker Cura 4.8.0I have an old Lotmaxx's Cura sliced gcode file that works fine.
What I have found out after some searching, a friend and I started to look at the "g-code that Ultimaker Cura and Lotmaxx Cura 4.8.0 slicer makes. We found out that the new 4.8.0 Cura versions are not adding the "L" in the "G28 L;Home".With the  "L" my Lotmaxx will print the new sliced filed great..
I have attached a comparison of the original Lotmaxx Cura that I have an old working / printable file / gcode. The one on the LEFT is the good file , and the one on the RIGHT is the problem gcode generated by Ultimaker Cura .
If you notice the "G28 L;Home" on the Left was the correction we made the correction / change and works.If you notice the "G28  ;Home" on the Right is a bad / will raise the nozzle up.
Dose this make sense ? Thanks, Albert Drew

rtyr, 2021-04-18:
It is unfortunatelly not possible to upload files via e-mail attachements.

Anyway, it seems resolved, so we can close this issue. That "L" parameter is likely specific to Lotmaxx's firmware. Placing that `G28 L` line to `Printer settings -> Custom g-codes -> Start g-code` should be enough (based on your findings).

closed 2021-04-18 by rtyr