prusa3d/PrusaSlicer #6377: Prevent 3mf files from overwriting filament/printer settings

WoLpH, 2021-04-18:
### Version
2.3.0

### Operating system type + version
OS X 10.15.7 and Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa MK3/S + MMU2S

### Behavior
When opening an 3mf file my filament choices and colours are overridden and I need to select the colour/filament choices again. Additionally it will override the selected printer which means you need to reselect this every time before you can use print to octoprint.

1. Select your own printer and filament in prusaslicer
2. Close prusaslicer
3. Open prusaslicer again, you should still have the same filament and printer selected
4. Open a random 3mf file
5. Your printer and filament choices have been replaced

I would prefer if prusaslicer would load the configuration options and not select them, or offer a dialog with the question to transfer the settings

#### Project File (.3MF) where problem occurs

[slider-left.zip](https://github.com/prusa3d/PrusaSlicer/files/6330287/slider-left.zip)

rtyr, 2021-04-18:
It is already possible. You can open 3mf without configuration.

[image]

Closing.

WoLpH, 2021-04-18:
@rtyr but that doesn't work when opening Prusaslicer from a 3mf file so this feature request is still valid. Once you do that, it has overwritten your settings and you need to configure it again.

Should I create a screencast to illustrate?

WoLpH, 2021-04-18:
At the very least... overwriting your settings by default without asking is bad practice and I would consider it a bug

bubnikv, 2021-04-19:
> Should I create a screencast to illustrate?

I suppose neither me nor Roman are understanding your issue. You are
questioning software behavior that nobody has an issue with for years.

WoLpH, 2021-04-19:
Screencasting is giving me a bit of trouble so I'll try to illustrate using a few screenshots.

1. My regular Prusaslicer state: 
[image]
2. **Close** Prusaslicer
3. From the Explorer/Finder doubleclick on the 3mf file I added as an attachment earlier, now the selected filament and printer have been overwritten by the 3mf file as can be seen in this screenshot:
[image]

Since the filaments contain the current colours for my MMU2S setup I need to change them back to see the expected result.
When I switch back to "my" printer and choose to discard all settings changed by the 3mf file, it selects the printer but doesn't select the right filament:

[image]

joesanford, 2021-04-20:
The purpose of 3mf files is to store additional information in addition to the geometry: https://help.prusa3d.com/en/article/saving-projects-as-3mf_1773/

This is why the "Import" functionality exists, in the case where you only want the geometry. However, this is opening and using 3mf files as they are designed.

WoLpH, 2021-04-20:
I understand the reasoning but I believe it's a bad default because it makes you lose your configuration. In my case I need to go to the printer again to see which filament is loaded in which slot, that information is lost. 

At the very least it would be very useful if it would ask or it would be a configurable option.

Can you explain in which case it would be a useful default to overwrite your printer settings? I understand that overwriting print settings is useful, but both filament and printer settings are generally personal to the user.

joesanford, 2021-04-20:
Lose what configuration? You can just discard changes if you accidentally load a 3mf. It's not a permanent override, and can be fixed by just reverting which printer/filament/print settings. If for some reason that isn't working for you with the MMU, then raise a bug regarding MMU settings. As it is, any of the settings can be easily reverted.

Making it a configurable option doesn't make sense and is then a breaking change in the spec and function of a 3mf file.

For a use case? Any time I want more than just the geometry of a file, or want to share more. Maybe see what changes were made for a trickier print, or for sending a more complex file over to someone I know who may not be able to set it up correctly to slice, but want to be able to see what was changed.

IMO, personal preference should not trump file specs.

WoLpH, 2021-04-20:
> Lose what configuration? You can just discard changes if you accidentally load a 3mf. It's not a permanent override, and can be fixed by just reverting which printer/filament/print settings.

Perhaps there is something wrong with my PrusaSlicer setup so please do correct me if I'm wrong here.

By default, I have a **selection** of several colours of filament set up in PrusaSlicer. This is the state I want to keep as the default because I have to go to the printer otherwise to see which slot has which type of filament loaded.

So here are my steps, feel free to tell me what I am doing wrong:
1. I open a 3mf file from the file explorer. My **filament choices** are now replaced.
2. I close PrusaSlicer without saving any settings
3. I open PrusaSlicer again. My **filament selection** has not been restored and I need to go to my printer to see which one is loaded in which slot.

An alternative would be to open PrusaSlicer first and import the 3mf file after that. But I think (correct me if I'm wrong) that I only get the model in that case. No option to load the print settings, filament choices or printer settings.

> Making it a configurable option doesn't make sense and is then a breaking change in the spec and function of a 3mf file.

I have to disagree because they have different scopes.
Print settings are strongly related to a specific model.
Filament settings and choices depend on what filament the user has loaded in their printer and only relate to the model slightly.
Printer settings mostly depend on the printer a user has and also has little affinity to the model.

I would argue that for most users (especially the ones using octoprint) it makes very little sense to switch the printer by default because they would need to switch the printer back every time they open a 3mf file.

> For a use case? Any time I want more than just the geometry of a file, or want to share more. Maybe see what changes were made for a trickier print, or for sending a more complex file over to [...]

That is a fair but rare use-case. A much more common use-case is to download a print from https://www.prusaprinters.org/prints and printing it on your own printer with your own filament. And the person uploading the file does not know what printer he/she is using or what filament he/she has.

> IMO, personal preference should not trump file specs.

I agree, which is why I'm opting for a choice :)

neophyl, 2021-04-21:
Yes you are misunderstanding how PS works and the load options you have.

You can think of a 3mf file as a project file.  it is designed to store the settings for the entire environment.  This is by design, its specified to do so.  As such it stores the filament selections etc too.  If a 3mf is saved from PS that's what it does.
Its also incredibly useful for debugging issues as well as a few other things.  
There are 2 options when opening a saved from PS 3mf file.
You can use File>import 3mf which will just load in the geometry from a saved 3mf and keeps whatever settings PS currently has loaded.
OR you can use File>Open Project.  Opening a project loads in the entire 3mf, including the geometry but also all the profile settings, modifiers, variable layer heights etc. Its this option that double clicking a 3mf file uses to open PS (if associated).  This is the same as any other windows program.  You wouldn't expect Word to open up a word document with just the text would you ?  You expect and get all the formatting too.  As such its currently behaviour is the correct and widely accepted one.

For example as PS doesn't as yet allow you to save pre-set workspaces from within the software its very easy to set up the environment for printer x, filament y and then save it as a normal project 3mf file,  I can then load that 3mf back in and get back to the exact settings I like easily.  So even with one printer I can load say my Petg setup by loading petg_blank.3mf.  You could do the same with your MMU settings.  Just save a project with your normal filaments and then you can later Open PS either normally or by double clicking on your saved project and then use the Import 3mf option to load in geo without changing the filaments you have set up.

Other software that also stores things in the 3mf format will also store their settings within the 3mf, although design based software wont have settings for filaments etc stored within them.  So when you use File>Open Project with those they dont replace the filaments as there's nothing to replace them with.

You seem to want the ability to save a 3mf from within PS and NOT save specific parts of the software settings.  

Btw the use case joesandford puts forward is far far more widely used than you think.  Its used multiple times daily on the Prusa forum to help users with their problems.

WoLpH, 2021-04-21:
> This is the same as any other windows program. You wouldn't expect Word to open up a word document with just the text would you ? You expect and get all the formatting too. As such its currently behaviour is the correct and widely accepted one.

When I open a Word file made by an American I expect the paper size to change to `US Letter`. But I don't expect Word to change the **default paper size** for any document in the future to `US Letter`. And it doesn't. The next time I open Word it defaults to A4 paper size as expected.

Additionally, Word won't change my default printer or print settings either when opening a Word file.

> You seem to want the ability to save a 3mf from within PS and NOT save specific parts of the software settings.

Exactly. I don't expect my **default settings** to change because I'm opening someone else his file.

> Btw the use case joesandford puts forward is far far more widely used than you think. Its used multiple times daily on the Prusa forum to help users with their problems.

Quite possibly, but I would hope that the default state for most users isn't solving problems with their prints.

In any case, I have a few questions:

1. Can we agree that my **selected default settings** are lost/overwritten when opening a 3mf file?
2. How could I open a 3mf file including print settings without overriding my filament/printer selection?

neophyl, 2021-04-21:
1. Yes they are.  This is by design.  However it only does this as PS remembers the last used settings.  Also by design.  There are no Default settings.  Just last used ones.
2. Open Prusa Slicer (this will open it with the last used settings).  Select File>Import 3mf.  This will import the geo from the 3mf WITHOUT changing any of the settings currently active in PS.

As I mentioned previously if you save a project with your preferred settings you can open that project later.  That will replace all the settings with the ones from the project.  As these are ones that you like when you saved it your work environment is then how you want it.  At that point PS has the settings you want loaded in.  Now use the Import 3mf option and NOT the load project option.  The geo will be loaded but the settings wont be.

When you close PS down the it should save the currently selected profile choices.  Those profiles selected should be exactly as you left them when you next open it.

This whole thing could be better addressed by making a feature request for something along the lines of "Allow users to Save one or more default workspaces and have the option to always load one of them at program start".  In that manner it would automate the loading in of a saved project to put PS into a 'known state' which the work around I describe currently allows.

WoLpH, 2021-04-21:
> Possibly not the best analogy but the word one was the first that sprung to mind. However Blender also does the same thing. You load a blend file in from someone else and it changes the programs entire environment and settings to how they are stored in the blend. You have to do a similar thing to load a file and preserve your working environment.

I think it was a good analogy actually, and it shows how the behaviour is non-standard. I personally can't think of any other software that changes your default settings by opening a file.

Even if Blender does the same (I rarely use Blender so I don't know how to verify), I still think it is strange to change someone's defaults without asking.

> 2. Open Prusa Slicer (this will open it with the last used settings).  Select File>Import 3mf.  This will import the geo from the 3mf WITHOUT changing any of the settings currently active in PS.

So I guess it is not possible? That doesn't include the print settings, I would like to load the model and the print settings from a 3mf file but outside of debugging I don't see the use for ever loading the filament choices from 3mf file if you only have a single printer.

> As I mentioned previously if you save a project with your preferred settings you can open that project later. That will replace all the settings with the ones from the project. As these are ones [...]

That is a nice workaround, but I still think the default behaviour is strange. Perhaps this is something we can somehow poll among users?

As an end user I think the current behaviour is inconvenient.
As a developer or someone that helps other debug issues with their printer it makes sense.

## Solutions

To shift this question towards a more productive direction, I can see several changes that could improve the user experience here.

These are not mutually exclusive, I think several of these would be great features to have:

- I believe it would be beneficial to link filament choices to a (physical) printer in the settings.
    That way you still need to reselect the printer every time you open a 3mf file but at least you don't have to reselect the printer _and_  the 5 filaments.
- When importing a 3mf file, it would be useful if it would load but not select the settings so you could still choose the settings if needed.
- Have a setting to not change your choices (but still load them) when opening a 3mf file
- Or have a dialog when opening a 3mf file that asks you which settings you want to load if they are different from your current defaults.

WoLpH, 2021-04-21:
Another few solutions:

 - Ask during exit whether to save the default selection
 - Only change the defaults when they were explicitly chosen by the user instead of loaded through a file

bubnikv, 2021-04-21:
@WoLpH 

Frankly you seem to be the only person that thinks current behavior is flawed. I would wait for others to chime in.

neophyl, 2021-04-21:
You seem to think that Prusa Slicer has 'Default' settings.  It doesn't.  It has **last used** settings.  

As I mentioned previously your request would be met by adding the feature for Default Settings to Prusa Slicer.  While I think that would be a useful thing to add I don't think its high priority as you can load in a 3mf project with settings you like.  Ok thats a couple more clicks but it works.

WoLpH, 2021-04-21:
@bubnikv That seems to be the way over here and I think there are several reasons for that.

1. Because I have an MMU the issue is multiplied. For a user without an MMU it only requires re-selecting the printer and 1 filament. Instead of the printer and 5 filaments.
2. The 3mf format isn't that widely used yet. Most models you can find on the Prusa prints site, Thingiverse and other websites are in STL format so many people might not have encountered the issue yet.
3. Many of the people here on Github are developers and reason from a developer standpoint instead of an end-user standpoint.

@neophyl That is only a naming difference, whether you call it "default settings" or "last used settings" makes no difference. The point remains that it is overwriting my preferences without asking.

What do you think of the solution of linking the (physical) printer to the filaments? That would make it far less annoying to use.

neophyl, 2021-04-21:
BTW end user here, not a developer.

No it is not just a naming difference.  Its a functional difference at a basic level.  Default implies that it goes back to a certain set of defined settings.  Prusa Slicer does NOT and has never (afaik) done that.  

I get that you have an MMU but at this point I am still struggling with what you are having a problem with.  Save a project with your filament choices and when you load it they will be set.  Take advantage of the fact that loading in a PS saved 3mf file does load in those filaments.

WoLpH, 2021-04-21:
> No it is not just a naming difference. Its a functional difference at a basic level. Default implies that it goes back to a certain set of defined settings. Prusa Slicer does NOT and has never (afaik) done that.

The point is that it is overwriting my choices causing loss of data which is a terrible user experience. Unless I had previously saved a 3mf file with those choices I cannot recover them.

> I get that you have an MMU but at this point I am still struggling with what you are having a problem with. Save a project with your filament choices and when you load it they will be set. Take advantage of the fact that loading in a PS saved 3mf file does load in those filaments.

That is a workaround, not a real solution.

Let me sketch a scenario that I believe is likely to happen for a new user of prusaslicer.

1. They install and setup prusaslicer for their printer.
2. They add and configure their filament
3. They configure prusaslicer to print directly to octoprint.
4. They download a 3mf file somewhere
5. They open the 3mf file and wonder why the octoprint button has suddenly disappeared and what happened to their filament since it's not in the list anymore because those are linked to the printer.

Is that a good user experience?
Since the filament dropdown is above the printer selection it is likely that they will click that first and don't know what happened to the filament.

joesanford, 2021-04-23:
> The point is that it is overwriting my choices causing loss of data which is a terrible user experience. Unless I had previously saved a 3mf file with those choices I cannot recover them.

Yes, and? This is how PrusaSlicer has worked since it has existed. You keep thinking about "defaults" but there is no notion of that in PrusaSlicer, just "last used". Since that is the case, _save what you want as a "default"_.

> That is a workaround, not a real solution.

No, it's a solution, as many of us have pointed out your concern has never been raised before, nor does anyone agree these are common uses cases.

> Let me sketch a scenario that I believe is likely to happen for a new user of prusaslicer.

This isn't even in the control of PrusaSlicer, as it is _treating the file as it is defined by spec_. This is not Prusa's spec, this is a widely agreed upon and used spec. To have an application treat the file differently because you don't like how it works is _not_ a valid reason to break from a spec.

Also, since you seem to think this is only collaborators/devs here: I'm just a user, who is a software dev. So, I'm just aware of how bad a practice it is to deviate away from an expected behavior of a file. 

Please familiarize yourself with this, and perhaps it will make it a lot clearer how much of a deviation from the spec this is: https://github.com/3MFConsortium/spec_core/blob/master/3MF%20Core%20Specification.md

WoLpH, 2021-04-23:
> Yes, and? This is how PrusaSlicer has worked since it has existed. You keep thinking about "defaults" but there is no notion of that in PrusaSlicer, just "last used". Since that is the case, _save [...]

Please allow me to repeat myself:
"causing loss of data which is a terrible user experience."

Just because it's always been that way doesn't mean it's a good idea.

That is the point of a feature request, to request a feature to improve our lives. Humanity has survived for millenia without vaccines, doesn't mean they're a bad idea. 

> > That is a workaround, not a real solution.
> 
> No, it's a solution

It's broken if you accidently double click a 3mf file in your explorer so it is by definition not a solution. Please look at the definition of a workaround. 

> This isn't even in the control of PrusaSlicer, as it is _treating the file as it is defined by spec_. This is not Prusa's spec, this is a widely agreed upon and used spec. To have an application [...]

I have read the spec and it doesn't mention anywhere that a dialog is not allowed. It doesn't mention that it is not allowed to give the user the choice of selectively importing the file. So please, quote me the part of the spec that my proposal is violating because I cannot find it. 

> Also, since you seem to think this is only collaborators/devs here: I'm just a user, who is a software dev. So, I'm just aware of how bad a practice it is to deviate away from an expected behavior [...]

Once again. Please read my proposed solutions and tell me how they violate the spec.

You appear to be arguing with my proposed solutions without actually looking at them... quite offensive honestly. I'm trying to make PrusaSlicer a better tool for everyone and I believe that the current implementation is confusing for new users and can easily be improved.

joesanford, 2021-04-24:
I had written up a response, but it's clear you just want to argue (including fallacies of personal attacks?), so I'll let the developers take it from here. People will disagree with you, no need to devolve into this.

WoLpH, 2021-04-24:
> I had written up a response, but it's clear you just want to argue (including fallacies of personal attacks?), so I'll let the developers take it from here. People will disagree with you, no need to devolve into this.

Please just take a look at the proposed improvements. I think at least a few of them are beneficial for all users and scenarios.

bubnikv, 2021-06-03:
@WoLpH Sorry, but since 24th April nobody chimed in and you are the only person known to us having an issue with the current behavior.

Closing.

ericvenneker, 2021-08-15:
My apologies, i know i am late, but I did not own a Prusa printer in April. I understand why opening a 3mf works this way, but I agree with WoLpH that it currently can throw you under the bus.

This 3MF file is created with a Mini in mind and a .25 nozzle. I own a MK3S+ with a .4 nozzle not a Mini with a .25 nozzle. I was not paying enough attention to the changed build plate because the printer selection still says MK3S+. I just changed the filament type and try to print (with some delicious prusament spaghetti as a result). 

Neither prusaslicer or the MK3S+ prevented the print. So there is either no nozzle check and printer type validaition or it did not kick in for unknown reason.

It is one thing to (temporarily) overwrite the current workspace settings. It is quite another to select invalid settings because 'the file said so'.

The Word analogy is quite apt, because it also contains some printer settings which are copied from the default template (normal.dot) on creation and can ruin your day if you reconfigure your printer. (I had to change several thousand word documents for a lawyer firm once). Both should not occur in my opinion because the printer is something specific for the user.

That said, you will probably not change this (only 2 'complainers' ;-)) so I made a mental not to NEVER open a 3mf with prusaslicer but only import them.

[image]

rtyr, 2021-08-15:
@ericvenneker The scenario you mentioned will be handled differently in the upcoming PrusaSlicer 2.4.0-alpha1. In this particular case, PrusaSlicer will temporarily install Prusa MINI profiles (so there will be Prusa MINI 0.25 nozzle in the printer selection). User will be notified about it. 
As for printer/nozzle checks on the printer side, MINI firmware doesn´t have such functionality yet, but maybe we can add those checks to the start g-code anyway (it would prevent using MINI g-codes on MK2.5/MK3 printers).

ericvenneker, 2021-08-15:
@rtyr Wow, that was quick. What are you doing here on a sunday evening!

Thank you for the heads up on the  new version. Funny thing is, i do not own a mini. But the 3MF was for a Mini. My MK3S+ however did not complain. My printer was set to only warn on an incompatible nozzle or printer type. I have set it to strict now ( I do not want a repeat ;-)) but with the same g-code it again did not warn me.

I have checked the the gcode file and I do not see an m16 check (I do not know if it is used in the firmware) but the comments at the end do contain the printer and nozzle type.

I have included the g-code and the 3mf file because I am very curious if I have made a mistake.
[prusa font 3mf and gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/6988858/prusa.font.3mf.and.gcode.zip)

screwyluie, 2021-12-15:
I agree with wolph as well. This is one of the most annoying behaviors of PS. 
I also disagree with telling someone their opinions and ideas don't matter or hold merit simply because no one else commented on their github issue and you disagree with them.

This overriding of last used settings is _ridiculously_ annoying and there is absolutely no reason you shouldn't be able to selectively use/review the settings applied when opening a 3mf file. **Especially** the filament and printer configurations which could have dangerous changes made in them and have little to no use when sharing with another person.

This should be easy enough to do with the new dialog that 2.4 has when you change profiles and it shows you the what's changed and asks if you want to keep the changes/etc.

open a 3mf, that dialogue opens up and shows you what it's going to change, make it easy to disable entire profiles (ie: check box for printer config, filament config, print config) so you can skip whole config sections. This would not only serve as a warning to unsuspecting users who don't understand what a 3mf is and that is changed settings, it also solves this issue presented here, and it allows people who do understand what 3mf is and like what it does to easily review any changed settings and make sure they're not malicious.

nathanielhudson, 2022-01-03:
I'd also like to chime in - this is super irritating from an end-user perspective, to the point that I will download STL files over the newer 3MF ones if they're available to avoid having to change back my settings every time.

maverickccie, 2022-01-04:
This is still a problem in 2.4. I fully agree with screwyluie and woph, this is hands down the most annoying behaviors of PS. Especially when downloading files from prusaprinters.org! I may not be using the same printer(sometimes even dare I say not a Prusa printer)/filament/nozzle as the person who uploaded the file. My process with STL's has always been to drag and drop them form the file explorer to PS. I made the mistake of doing this with a 3mf, the little pop at the bottom is too little too late, by the time the file was in all my settings are gone. So, the fix is I have to do a manual import? This is not good workflow. Why can't there just be an option that says verify drastic settings changes before allowing them or on a drag and drop just import the model? This is why I stick to uploading STL's, I would never assume to push my settings on someone else.

nathanielhudson, 2022-01-17:
Since comments are the metric being used here, I would also to say that I agree with screwyluie and maverickccie. The default (as in, most straightforward user path - "drag file into program") behavior being "assume that I use the same printer as the person on the internet, and I want to use their printer's settings for some reason" feels very odd. I deliberately download STLs rather than 3MFs for the above reason.

foltak, 2022-03-21:
I would also prefer if it didn't overwrite filament and printer settings or if you could choose which settings to include. When I'm uploading files to prusaprinters and I create custom settings in the project that are important for the person printing the files to have set correctly I don't want to change their printer type or filament settings without them knowing. So now I have to export them for individual printers so someone less experienced doesn't accidentally use the wrong settings.

traderhut, 2022-03-22:
> At the very least... overwriting your settings by default without asking is bad practice and I would consider it a bug

Agree 100%!!!  I would NEVER have expected that if I open an old file that I'm going to have to set everything up again to print it on my new printer...   I've quit using this program multiple times because it is just so dang annoying to try to use it - because of THIS issue.

Saying that instead of 'open recent project > Foo' you can open a blank one, and import the project the project to not get the settings is ROFLMAO funny...   Yeah, there is a work-around, but seriously?  That isn't 'the feature exists'.. That is almost like saying you can edit the file with a text editor and remove all the settings so we don't need the feature.

It is nice to know there is a quicker way to do this than the huge hassle of trying to reset all the settings - which 'undoing' the changes should be:  Click, Confirm, Done!  I have spent 20 minutes trying to figure some way to get the settings undone, and just constantly fighting with the UI to try to get it back to what I had setup on the last file I printed.

foreachthing, 2022-03-22:
> @bubnikv 
> Frankly you seem to be the only person that thinks current behavior is flawed. I would wait for others to chime in.

Not flawed, it has room for improvement. 😉 

My 2 cents: I just want to get asked (or, preferably have permanent settings/preferences) whether I want to import/activate the settings from the 3mf or not.
I don't like how PrusaSlicer currently handles double-clicking 3mf files either. At least make it **optional** - let the user decide.

My workflow currently is, always to open PS _first_ then open the 3mf. It's not terrible once you know it works, but it was (unnecessarily) confusing at first and would make _my_ life easier.

traderhut, 2022-03-22:
BTW - the comment that Blender does the same thing - it make sense in blender, you might have a bunch of stuff that is project specific that needs to be setup for a graphic design application.      How often would you open blender file and have the project's results in ANY way be changed by ANYTHING external to the application?  You wouldn't it will be generating digital output that feeds into other apps (I don't use the app).   If you changed the nozzle on your printer that wouldn't make any difference to Blender, if you changed anything it wouldn't.   I don't know if it remembers what printer you had selected, but I can imagine people would hate that if it did.   Consider a work environment, (I used to work at a Game Studio that generated graphics for games that were 3D.)   OK, let's say they added a new printer and told everyone in the Graphics department to use this printer instead of the (more expensive) printer for all prints.   No problem, you change your Windows Default Printer.  (or other OS), and just open the files and print.  You might have 10,000+ files saved on your machine.

Now, what if you had to update each one of those when you opened them?

I get that you have no default settings, and maybe it is a feature to make it hard to switch to the last used settings, but honestly, in 40+ years of using software as a software architect, I've NEVER encountered a program that was designed this way without a use case for it.   - How often is a "OMG I'm SO glad that it restored that setting when I opened the file, because otherwise I would have totally printed this wrong!"  Vs "OMG! I forgot to update my settings, and I just wasted $5 worth of material!"     (I actually have a HUGE issue with what it just did... I have custom GCode that is run at the END, this code was brought in from a previous "last used" setting when I was using a different printer, specifically, it did a 'G28 X0 ; Home X Axis' which would move the bed out and make it easier to pick up the part.

Great.   That custom Code is stored with every file I've ever worked on with the old printer.
Now, I open that in my Tool Changer, which (currently) has an issue that I can't print the first few locations on the X axis -as it is hitting the side of the machine if I attempt to use that code.  And when I say hitting, well, hitting HARD.  The device will auto detect that it wasn't able to move there, and un-home all 4 Axis's, It will then not execute the T-1 (Select no tool) to part the tool back where it belongs, and it will forget that it has a tool.

Not only could this damage my printer, but when I go to print something and it homes again, it will drop the current tool before doing that (better be there to grab it, and when you do be careful, it could be quite hot!)

Now, every time I print a file that was printed with the old printer, I risk that I didn't either specifically go off and load settings for the new printer... And I might damage my printer!

Also, I had my printer bed set wrong at one point for size, so I fixed that... and the software auto-unfixed it when I just loaded a file...  BAM!  Printer rams into itself and E-Stops (some printers won't detect a critical failure due to a mis-configuration like this and will keep trying to print with the location now shifted.. Been there, done that..)

If I had more than one printer (I have 5 3D printers, a CNC router, a CNC Mill, a CNC Engraver, a CNC painter), then I might want to switch things often, and I sure wouldn't (very often) want to have all my printers defined in my print files, but rather would like to set them up in the software...

I can't be the only person who has more than one printer...  Or who uses custom GCode that could cause major issues on other printers!

And if you load up a file, you see NO indication that 3 of the setting tabs have changes from (won't call it a default) what the printer NEEDS to be able to print correctly.

(So, anyway, in reference to the preceding, I am clearly not the only person who has this issue - I think it is storing information about something outside of where it make sense to store it because you are familiar with another application that does it, ignoring the 1000's of applications that don't)

traderhut, 2022-03-22:
BTW: The issue with the printer slamming into the side when it ran bad Gcode... It was not a contrived example, it happened to me today, with my 2nd to my last print, and I had to re-setup the printer following it and try to figure out how the heck did that code get back IN my gcode file???

Thank goodness I wasn't trying to drive something that could be more seriously affected by it.

traderhut, 2022-03-28:
After thinking about it some more, I think that the solution is to move printer/machine specific settings out, and leave project specific settings in.   For example:    Is there a Raft, if so, how big?   Project specific.   What is the size of the print bed?  Printer specific - doesn't belong in a project file.  You wouldn't want to open a project that you created on a large printer (or one someone else sent you), and have it think the bed was 300x300, when you had a 150x200 bed - and have it render it all wrong, and ultimately not print it, wasting materials or worse.  Or have it switch from sending the job to a different printer, which someone has a low temp plastic loaded into and you just sent it a job at 285C... and it is damaging the printer...

Other settings that make no sense in a project file:   

- Gcode flavor (Marlin etc.)
- G-Code thumbnails
- supports stealth mode
- supports remaining times
- use relative E Distance
- # of extruders
- Single extruder, Multi Material

I hope that you get the idea, and understand why I don't think these belong with the project.

neophyl, 2022-03-29:
I completely disagree.  Those details might not make sense to you, but you aren't the only user of the software.  Different people use the software in different ways.  I want my projects to be a complete snapshot.  I expect them to generate the same output when loaded and sliced with the same versions of the software.  The whole point of a project is its supposed to be a self contained entity.  If I want a different printer or filament then its just a few seconds to select different profiles.  Frankly if you are opening someone else's project without at least a cursory inspection before slicing then you are just asking for trouble.  Its almost as bad as downloading gcode from the internet and running it.  I hope you don't download software from the internet and just run that too without checking it.

Its ok giving more options on the opening of projects but all that information needs to remain as part of the project.  

Also one aspect that it looks like you haven't taken into account is that the project mechanism is also the primary avenue for help.  Anyone trying to help someone needs ALL the settings.  A printers start gcode for example makes a big difference. I have helped MANY users over on the Prusa forum.  Most of that wouldn't have been possible without having a complete picture.  I am not the only one either.  All the regulars ask for a saved project file.  Same as the Dev's here when you open an issue.  It is a requirement.

WoLpH, 2022-03-29:
As I have mentioned before, I don't think the biggest issue is that the project files contain all of that information. The biggest issue is that it's overwriting your settings without asking.
Any feature that causes loss of configuration without a big fat question in advance is a terrible idea in my opinion.

The new preset comparison feature is very nice to see what has changed. Unfortunately it appears read-only for the time being so it's not all that useful in partially applying settings from a file.

With regards to helping people, that's definitely a power-user feature that should be less common than downloading and running a file from the 3D models database. I can appreciate having this option available but it should not be the default.

traderhut, 2022-03-30:
OK, from a support standpoint, it makes sense to wrap up EVERYTHING in the project file.  I get that.  On the other hand, when you get a project and want to print it, are you ever saying "Oh, thank goodness it set my origin to -2,-47 and my bed size to 300x150 for me"   (The E3D tool changer seems to have a very strange origin, which frankly I think is a bad idea, but I'm having that discussion with them - you could rewrite all the config files to reset it to 0,0, but they didn't)

The issue I had was you refer to it being just a few seconds to fix the settings.  I presume, you are suggesting that I select some file that I know has the correct settings and do an import settings from that file to get them,  In the UI I never figured out how to get them reset to what I always thought was the default, which you are calling the last used.  If it was an easy (and obvious) one click to reset them to those values, I'd have never even had a problem.  However, I've looked at the screen, saw settings that had changed (from what? who knows), and no way to CLICK and boom reset.  

I never looked in the 'file' menu for any import settings, it simply never occurred to me that I'd be resetting my settings for a project by using *anything* in the file menu.   It wasn't until someone (in this forum) mentioned that I could simply do that, that I knew it existed,  but then, I've only used the software since about 3 years ago, and installed it three times (uninstalled it twice).  My point here is that it is very unintuitive.  If you want to change your preferences/settings and you have 3 tabs of settings.. You go there, NOT to the file/ menu.

But, in reference to your excellent example of why it would be weird for me to expect the settings in the file... OK, so I download an stl file from the internet, drop it on my platter... and it changes my bed size to 60x75, temp to print.. extrusion speed, etc?  Yikes!  No, I just wanted the model dropped on the project...  Oh, well good thing it works that way...

Would it be nice if they saved their last successful print speed, and settings in the model and allowed me to set those options when I dropped the STL file on my platter to print it?  Yeah, if it prompted me first, that could be cool.  By default, behind the scene?  OMG No.  I don't want some random settings when I've been working so hard to dial in my printer...

The primary point is, it should be easy to add something to ask.

I like the idea of showing the differences, and you can select Apply, or keep (Ideally on each one) - that would be awesome!  Because, I would know if I wanted to reset the size of the bed, or change the amount of the retraction.

It would also be awesome (I would think) for your support people, when you open up a file and it says that the max print speed is 30000 mm/sec and you can say... AH, I expect THAT is your problem...  (The E3D has stuff in there that actually does F50000, but it is limited in other places from what I can tell)

So, anyway, My preference:   Dialog saying "About to apply the following changes to your settings... Would you like me to?"  - Which you already have code that does that when you save your settings, so it shouldn't be too hard to make those changes.
(And you could have it save a preference for if you wanted to be prompted or not)

(If I could get the app to compile, I could even do that as a Pull Request, but it won't, but that is another ticket...)

foreachthing, 2022-03-31:
My opinion: the way it currently works should be reversed:
* double click on file should do an "import" and only load the model (not changing nor importing any settings!).
* if you want to load the settings from the .3mf, then you would have to do it via files->import settings from file... or something.

I have my settings and please leave those alone. I don't even want to use **my** settings from a half a year ago...

If I download a file (3mf) I just want the model - not the settings (couldn't care less for those)!

Another thought: why don't you boys/girls at Prusa have a custom debugging-Slicer? That should be an easy one for you....

neddy, 2022-07-14:
+1 for wanting a preference to disable the loading of print settings when importing a 3mf file.

jkavalik, 2022-07-14:
I agree with the OP here too. There is a reason I do not actually have 3mf assigned to PS on doubleclick.
(Mis)using the 3mf format sounds great as a "project file" and is almost perfect when just I use it, but when I start downloading models from the Internet then all that goes out the window.
We can probably agree that switching some colors is not a problem, but most of the settings are about actual physical things that are not at all the same between any two users.
Jo says that "Prusa*" tools are not just for Prusa printer owners - prusaprinters actually got renamed to get rid of that association but I believe the same to be true for PrusaSlicer - it is a great tool for many users with all the different Enders and Anets and many DIY printers and because they downloaded and opened a file saved by someone who owns a Mini, their user experience goes boom.
3MF is supposed to be THE upgrade over STL - smaller, correct etc, but randomly it throws at me a printer I never owned and possibly never will because I just liked the geometry and forgot to start the slicer by its desktop shortcut before opening the file.
I too believe that this issue is currently very "underrepresented" because 3mf is still not so prevalent and MMU si very specific tool. But the XL is supposed to bring more materials and even more disjoint configs between the different extruders and the idea that my most used presets set "vanishes" in a doubleclick and I have to configure it again, maybe even walking to the printer, because I don't remember if the 0.8 nozzle was E3 or E4? The second time that would happen I would probably scream :)

I understand that File->Open Project is expected to behave this way but I suggest to change the "app opened by file (click, argument)" to be changed - buttons offering buttons for "geometry only", "load as project" and possibly "diff to my current settings" or "load configs without selecting".

OTOH - on 3mf creation it gets hard to NOT include the settings even when I think about it and want to put clean geometry 3mf to printables, as we only have "export object/plate as STL" and no 3mf export option. I actually created some of my MMU uploads in meshmixer just to give give users a model with single file, multiple named parts and no config hassle.

neophyl, 2022-07-14:
I agree on changing the default action for double click and all the various options people want to have for choosing what is loaded from a 3mf file but the bit I don’t get in all these posts is the “I open a 3mf file from xxxx and I lose all my settings”. Doesn’t anyone use the save profile functionality ?  
Either as actual print/filament/printer profiles or simply just saving various default project files ?  

If I open a project from Printables or the PRUSA forum and it uses say an Anet A8 then it’s seconds to just select my printer from the drop down. Same with filaments and print settings. The mmu/multi filament settings are trickier admittedly but even there that’s where saving a few default 3mf files of your own comes in. 

Am I one of the few who also uses the export config bundle option to generate regular backups too ?  Just in case of computer problems. The tools are there to save your favourite setup. Make use of them.

ThisIsMyNameBecauseEverythingIsTaken, 2022-08-04:
This topic is the perfect example of the two biggest stereotypes in FOSS..

* If it ain't broken, break it (3mf changing active settings profile)
* The end user is always wrong (don't allow changing default behavior from import everything to geometry only)

neddy, 2022-08-05:
> I agree on changing the default action for double click and all the various options people want to have for choosing what is loaded from a 3mf file but the bit I don’t get in all these posts is the [...]

You're missing the point... It's not that you can't work around it, it's that it isn't intuitive... Normally 'import' would be used as a label for an action that imports settings etc... and open would just open the file without changing all the settings etc... For example, let's say I am trying out new print settings and minor tweaks, and then I want to load a different model to try the same settings with, I can't just drop the 3mf file into the open window with all my recent tweaks to the settings as I will lose them, instead, I have to import the file so that it only loads the geometry. I can definitely see the value in being able to save config/print settings to the 3mf file, I just think the default should be to load just the geometry. Even if this was just a tick box in preferences to only load geometry by default from 3mf files, that would be great.

I would also just like to add that I love PrusaSlicer, it is amazing! I am using the latest Alpha with the new Arachne perimeter generator, and it's awesome. 

By no means am I trying to dump on the developers that are doing an amazing job, I just wanted to add my voice to say that I think there is a good case to change the default as it would be more inline with what I think most people would expect 'import' and 'open' to do, or at least add an option to not load setting from 3mf files when opening them.

mattddonnelly, 2022-09-22:
My theory on the reason why no one had a problem with this before is because no one was putting 3mf files on e.g. Printables until things like this blog post from Prusa last year:
https://blog.prusa3d.com/3mf-file-format-and-why-its-great_30986/

Now everyone thinks it's great (valid for some of the reasons outlined) but it means that the behavior is likely now being relied on significantly more than it ever was before--and thus, people complaining about it.

I'm curious if PrusaSlicer metrics record the number of times people have imported 3mf files, and the method of import, as a part of understanding how people are using the application now compared to before?  Does that data confirm or refute that theory, that there's been an uptick in ~the last year or so?

This is (roughly) how I think it should behave, in line with many other posters here:

_All_ methods of import should behave the way OP wants: not overwriting settings (possibly loading them in the background but not replacing the presently loaded configs).  That goes for launching PS from a 3mf, drag-and-dropping a 3mf into the plater view, and File > Import > Import [types] options.

**Defaulting the behavior to load configs especially doesn't make sense when Printer is one of the configs that gets modified in PrusaSlicer.**  Why would I want someone else's printer profile when I download a model from Printables?  That's an exceedingly rare desire, relative to the normal case where people are downloading these in place of STLs.

See also: when dragging a 3mf in after dragging a prior 3mf in, it prompts whether I want to load geometry only, etc.  Treating the base plater view as its own "project" would also presumably just enforce this behavior and prompt every time.  Annoying, but not as much as starting a print only to find out I sliced for an Ender 3 instead of a Prusa Mini.

I would argue there should be an additional Import option, something like "Import Project (3mf, [...])" which loads all configs as well as geometry--or prompts

i-like-trains72, 2022-09-24:
+1 on mattddonelly's comment, sums up what I wanted to say.

Simply allowing us to toggle what settings are imported when importing 3mfs would be useful. 

Geometry, supports, overrides on specific parts of the model, are what we want with 3mfs being imported into PS. Printer settings, and filament settings, these are what we tune for our own personal situation and we don't want them overridden by someone else's 3mf.

kobaj, 2022-10-17:
+1 to WoLpH, screwyluie, mattddonnelly, etc. 

After spending all day printing, I realized that the 3mf file I had grabbed was using an extrusion width of 200% and override my settings. Measuring the prints found that sure enough, they are all out of spec and need to be reprinted. What a waste of time and pla! The default behavior should be to just import the object, not the print settings.

JohnOCFII, 2022-10-17:
PrusaSlicer has an excellent way to display modified configuration when changing between profiles.  It would be **great** if PrusaSlicer displayed something similar when opening a 3MF file that contains configuration items that are different than those currently selected.  That way, users who are expecting the opened 3MF file to bring along specific selections can confirm that action, and those who **only** want the model geometry can easily get that too.

[image]

wolph, 2023-01-15:
> @wolph
> 
> Frankly you seem to be the only person that thinks current behavior is flawed. I would wait for others to chime in.

Since 3mf is gaining traction this issue is starting to become more prevalent and this issue shows. Can you please re-evaluate this issue?

traderhut, 2023-01-16:
> @wolph
> 
> Frankly you seem to be the only person that thinks current behavior is flawed. I would wait for others to chime in.

Not sure if you already counted me as someone else who thinks that the current behavior is (dangerously) flawed, but sure, throw me in the list of 'others'... My solution to getting problems fixed currently is to download the source and fix them myself :-(.  I had to do that to support the E3D printer (multiple tools that move the z axis when you do a tool change, and have no way to do a 'move me back where you were, taking into consideration that the new tool might be off in x,y and z by some amount).

But, yes, I've stopped using this file format whenever possible, as it is too dangerous to open one and BOOM! all your settings are messed up, you can, of course track down a file you saved recently that had the right settings in order to restore it...

In fact, you could make it a habit of working around this defect, by having a 'MySettings.3mf' that you load up which has very little in it, maybe just a small dot/ring you can delete after loading it, this would allow you to restore everything in the event that you used a file someone else created (not that, there are any sites out there that offer pre-drawn 3D models that anyone might load),   This would allow a 'quick' work around... And then, any time you wanted to change your settings, you just do the following:

1. Save your current file.
2. Open the 'mySettings' file (which wipes out any changes you had to your current file)
3. Make the changes you want, and save it.
4. Load your original file. (wiping out your changes)
5. Load the 'MySettings' file to restore your new settings
6. And then test your new setting with your current file... If you want to try out another change, just repeat!  Easy peezy..

Or, of course, you could just answer 'no' to the prompt when asked if you wanted to load the settings for the file(s) that you downloaded from the internet....

Nah, that would be too much of effort to do for people...  (Which, of course, could be turned off by a setting in the software..)

traderhut, 2023-01-16:
PS: I won't comment on what 'MF' stands for in the extension, but I say it every time I get my settings wiped out...
Which, as I said, I'm not using that 'MF' format any more than I need to..  (Although, honestly, I think the concept of it is great!)

traderhut, 2023-01-22:
I would also add, that Yes, there is a faster way that I didn't think of until after I wrote the above, but you can, as you can load just the settings - there was some way to do it, I forgot how, but I think someone already mentioned in this ticket how to do that, so I won't try to look it up, but something like 'import settings' or something.  So, maybe we knock off a step or two..

julie777, 2023-01-27:
I will add a slightly different perspective, or maybe this is just distilling the above into a more concise wording.

**Normal Use Case - Open Project**

When I create a model in Fusion 360 I save it as a 3MF file. This way the model has units for the dimensions which makes it complete. I never use STL files because of the lack of units.
After I configure the model in PrusaSlicer and slice it I save the 3MF as my reference model and this is what I would upload to thangs.com for others. That way they have all the information to either print it the way I did or at least see what settings actually worked for me. (Ignore the possiblity that I store the gcode files for printing copies later.)

**Another Common Use Case - Import Object**

What if I want to add multiple models to the plate to print them together. In this case the models might have different settings and that needs to be handled.
In this case I want to import the model, not open the 3MF file and all its settings for everything.
There will be some metadata that I want and some I don't? (I want the units)
There will be some metadata that matches the current settings (which are from opening a previous 3MF, when adding another object, or default settings if the plate is empty.)
In this case it would be nice to be shown the difference in settings between current, and the model being imported. Then I could choose which to use.

**Conclusion**

Open Project 3MF should use all the information from the project.
Import Object from 3MF should show conflicting metadata and allow choosing which to use.

I believe that the differencing code for all the settings already exists, so this only requires adding a Choose Settings Dialog for the Import Object command.

lukasmatena, 2023-02-01:
Behavior was changed in [2.6.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2). Can we close this?

screwyluie, 2023-02-01:
I guess? I feel like the basic concern was addressed. I also feel like it could be handled better using the save settings comparison dialog. I suppose that gets shifted to a feature request now huh?

lukasmatena, 2023-02-01:
It would be cleaner to open a separate ticket for that, given how convoluted this one became.

wolph, 2023-02-12:
> Behavior was changed in [2.6.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2). Can we close this?

The new behaviour looks much better. I still think the default shouldn't be "Open as project", but at the very least we now have a way to work around the issue :)

closed 2023-02-12 by wolph

verne64, 2023-08-02:
As Neophyl mentioned, I've opened another issue which was caused by this "behavior change". 
This really bothers me, I'm trying to get used to first opening the software and the loading the file, but must say I'd much rather just double click the file in explorer. 
In my opinion this is a really bad way of solving this issue, resulting in an inconsistent user experience that forces me to have to think about which way I have to open a file (such a basic operation really shouldn't require any thought) and also having to explain this to people I'm sharing the file with (then wondering if they will remember how to do it, which they probably won't). 

I can relate to the points being made, but I've grown accustomed to using project files and not rely on profiles as every job is different and requires unique tweaks and I don't use an MMU. I feel it would be better to revert to the old way of loading project files and just add a functionality to let users choose their default profiles for when they start the software regularly, I think that would solve Wolph's issue as well right?

[6 more replies]