prusa3d/PrusaSlicer #6379: "Interface Pattern Spacing" shows wrong pop-up help when switching from Expert to Advanced

timur-tabi, 2021-04-18:
Version 2.3.0+linux-x64

Steps:

1. Go to Print Settings, click on Support Material
2. Switch to Expert mode.
3. Hover over "Interface Pattern Spacing" and wait a few seconds.
4. Notice that the pop-up help shows more info on "Interface Pattern Spacing".  This is expected.
5. Switch to Advanced mode.
6. Hover again over "Interface Pattern Spacing" and wait a few seconds.
7. Notice that the pop-up help now shows more info on "Pattern Angle", which is an Expert setting that is now hidden.

This problem does not happen when switching from Simple to Advanced mode.

The same thing happens with the "Elephant foot compensation" option.  I suspect there are others.

combolek, 2021-04-18:
FWIW, I cannot reproduce it, neither with 2.3.0 nor with the latest master.

timur-tabi, 2021-04-19:
It appears to only happen on Linux.  I can't reproduce it on my Mac.

combolek, 2021-04-19:
Yes, I am using Linux 2.3.0 AppImage and master build from source.

timur-tabi, 2021-04-19:
I restarted PrusaSlicer and I can't repro the problem either.  It must have been some strange side-effect of something else I did.  Oh well.

closed 2021-04-19 by timur-tabi