prusa3d/PrusaSlicer #6381: Slicing Errors in PrusaSlicer: large holes are missing after slicing

fex42, 2021-04-18:
Unfortunately I cannot re-open Ticket #6378 that I already closed.

The bug described in #6378 disappeared after trying the AppImage version so I thought it was a problem of my machine setup. But after a change in the model it reappeared (im AppImage and local installation) and also a forum user could reproduce the bug on a Windows installation with PrusaSlicer 2.3.0 with the new model attached here.

So except for the 3MF / STL project file, the description is a copy from ticket #6378.

### Version
2.3.0

### Operating system type + version
Arch Linux (I suspect the error is operating system independent)

### 3D printer brand / version + firmware version (if known)
The problem is printer independent, you can use any printer to reproduce Bug.

### Behavior
* Description: When slicing halfway complex models, slilcer errors happen that are not visible in the model (but in the preview). Mostly areas are filled, which should be a hollow but also other errors occur.
Unfortunately, the problem occurs only from a certain complexity, I could not create a trivial model with this error. I have attached an example object as 3MF file where you can reproduce the problem.
* _Steps needed to reproduce the problem_: Slice the attached model, look at result in preview
* _Expected Results_: Correct slicing of the model as Cura and/or IdeaMaker does it, see screenshots below of them
* _Actual Results_: slliced model contains errors (the large hole is missing), see screenshots below 

If you need more example models that can reproduce this behaviour, I can add others too.

#### Project File (.3MF) where problem occurs

[Prusa-Slicer-Rebirth-Error.zip](https://github.com/prusa3d/PrusaSlicer/files/6332105/Prusa-Slicer-Rebirth-Error.zip)

just for completeness here's the STL file too that produces the bug:

[kp3s-superframe-v5.zip](https://github.com/prusa3d/PrusaSlicer/files/6332108/kp3s-superframe-v5.zip)

### Screenshots from *PrusaSlicer*

* Model:
[image]
* Preview (the large hole is missing and the three large round sockets are not sliced correct):
[image]

### Screenshots from *Cura*
* Model:
[image]
* Preview:
[image]

### Screenshots from *IdeaMaker*
* Model:
[image]
* Preview:
[image]

fex42, 2021-04-18:
As it turns out, it seems to be caused by a problem in the STL file. Importing the STL in Tinkercad and re-exporting it will not produce the error.

My broken STL was generated with OpenSCAD 2021.01 and generating it from the exact same source produces a correct STL.

Nevertheless PrusaSlicer should slice it correct like Cura and IdeaMaker does or it should complain about a broken STL but not silently slice garbage.

fex42, 2021-04-19:
The broken STL file came from OpenSCAD 2021.01 when it exports as "binary STL" (which is the default). 
A possible workaround is OpenSCAD is that you set "STL export in ASCII format" in the preferences.

For reference, here is the OpenSCAD ticket for the bug: [https://github.com/openscad/openscad/issues/3644](url)

Since the other Slicers (Cura, IdeaMaker) don't have a problem with the broken STL, I leave the ticket open.

rtyr, 2021-04-19:
Already fixed on master (9c5dd35), the fix will be included in the next PrusaSlicer release.

```
Fixing model import of models with nearly degenerate triangles, which
degenerate after transformation (translation or rotation).

There was certainly a bug in PrusaSlicer, however this bug was triggered
mostly by the ASCII STL files generated by OpenSCAD, likely due to insufficient
number of decimal digits when saving an ASCII STL, STL export should contain single precision floats
```

Closing.

closed 2021-04-19 by rtyr