prusa3d/PrusaSlicer #6382: Windows Standalone 2.3.0 install switches ignored

djb160, 2021-04-18:
### Version
2.3.0 Windows Standalone (PrusaSlicer_Win_standalone_2.3.0.exe)

### Operating system type + version
Win10 Pro 20H2

### Behavior
Trying to figure out SCCM deployment. Running Powershell "PrusaSlicer_Win_standalone_2.3.0.exe /?" to get install options.
For a most basic install it seems like I should be able to use "PrusaSlicer_Win_standalone_2.3.0.exe /exenoui /exenoupdates"

But it still seems to be putting me through the UI walkthrough.
I tried extracting the msi and running with basic /i /q switches but it complained about not being run via a bootstrapper.

Janys, 2021-04-19:
Can you please try running it with /qn, /quiet or /qr instead of /q?

e.g.

/exenoui /exenoupdates /qn /quiet
/exenoui /exenoupdates /qr /quiet

Janys, 2021-04-19:
You can find some more information here:
https://www.advancedinstaller.com/user-guide/qa-silent-msi-install.html

@rtyr Can you please close this issue? We can consider it resolved.

closed 2021-04-19 by rtyr