prusa3d/PrusaSlicer #6386: New Feature- Add buttons for rotate around Z: +90°,-90° 180° ( frequently used )

birtalanelod, 2021-04-19:
### Version
in upcoming version implement 

### Behavior
* _Describe the problem_
these exact transformations are frequently used, and it is annoying to 1. Click into text field 2. enter all the time 90 on the box and 3. hit enter, compass rotation is to complex for this simple transformation

_Is this a new feature request?_ YES

neophyl, 2021-04-20:
Why is the compass rotation too complex ? It has snap to positions for 90 degrees which makes this operation extremely simple to perform. Are you using those positions ?

neophyl, 2021-04-20:
You shouldn't have to force yourself to find out.  May I suggest then that you review the software knowledge base, specifically https://help.prusa3d.com/en/article/move-rotate-scale-tools_1914 which mentions the snapping and how it works.
Personally I find the rotate tool to be one of the best I've used.  

One of the problems with PS is that while originally there was a scarcity of instructions in its use, now that there is, not enough people are aware of that fact.  The existence of the knowledge base is not obvious and many users both here and on the forums are not aware they exist.  This causes far too many raised questions that if they knew about the instruction manual wouldn't have been raised in the first place.

Getting to the knowledge base take multiple steps, a link from help in the slicer, just called 'PrusaSlicer Website' ,  which takes you to the download page and from there another click on a link to the Documentation.    Lets face it if you are clicking a link in PS help you already have it installed and so probably don't need to download it so going to the download page isnt that helpful.

It really needs to be a single click from Help , possibly called Instruction Manual or something similar that takes you straight there.

birtalanelod, 2021-04-20:
Agree with both of you guys, after you being so insistent I also managed to figure out that I have to hold the mouse over the white scale for snap to occur. It is not intuitive, at least some color change on the compass would be great, to provide some feedback.
Also still believe some rotate left / right button or right-click menu item would be a ton faster.

tullo-x86, 2021-04-20:
As a software dev myself, let me remind you that users are pathologically, nay, _morally_ opposed to reading manuals. In the long run, the most effective order to prioritise discovery is:

1. **Affordance** — Make the feature intuitive or self-evident: e.g. a visual indicator when snapping, or reusing common shortcut keys. (This is the lowest-friction option)
2. **Tutorials / first-start walkthroughs** — Show the user how to do it, right where they can see it. (This can work, but make it too long and users will just skip their way past it and be lost anyway)
3. **Documentation** — Let's face it, nobody goes looking for the manual until they're well and truly lost. This is fine for something like ArchLinux, where it's pointless to be installing it if you don't really want to understand how all the pieces fit together, but for an application targeting a wide gamut of user experience, having the manual be the _only_ source of truth (it should still be the _ultimate_ source of truth!) does not foster a satisfying user experience.

I'm honestly trying not be sound like a jerk here. App design is a discipline unto itself (which is why UX is an entire field that some people specialise in). I only mean to throw what knowledge I have (scant as it is) up for consideration.

bubnikv, 2021-12-08:
rotation of objects by 45 degrees is easy with page up / down buttons and the behavior is documented in the keyboard shortcut page.

bubnikv, 2022-06-07:
@tullo-x86 

> rotation of objects by 45 degrees is easy with page up / down buttons and the behavior is documented in the keyboard shortcut page.

Select an object on the platter, then press page up or page down key to rotate the object by 45 degrees.

tullo-x86, 2022-06-08:
Yes, I saw that response. Did you read my comment before replying with it?

closed 2026-09-29 by prusaslicer-bot[bot]