prusa3d/PrusaSlicer #6391: (critical) Cannot slice anymore in 2.3.1

Moyrn, 2021-04-20:
### Version
_Version of PrusaSlicer used goes here_
2.3.1
_Use `About->About PrusaSlicer` for release versions_
2.3.1 Win64
_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Win10 1909 18363.1500

_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
MK3S+MMU2S (latest firmwares)

### Behavior
* _Describe the problem_
cannot slicer anymore. Takes forever saying "preparing infill"
was ok in 2.3.0 - same file
[MANCHON.zip](https://github.com/prusa3d/PrusaSlicer/files/6343661/MANCHON.zip)

* _Steps needed to reproduce the problem_
open m3f, 
click slice

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_
no
#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
see zip file

_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

lukasmatena, 2021-04-20:
I tried and it hangs in 2.3.0 as well. Disabling either of 'Combine infill every n layers' and 'Ensure vertical shell thickness' solves the problem.

Moyrn, 2021-04-20:
Hello

I believe you. Though I just opened the file in 2.3.1 to pursue my work and
it failed. I printed with this file in 2.3.0 this morning.
Maybe a mistake. Weird though.

I however cannot launch 2.3.0 standalone anymore to confirm.
"Internal error - invalid string position" at startup (no project loaded
yet)
[image: image.png]
then "unhandled unknown exception terminating the app"
[image: image.png]

Oh my... yet another of these days

Wilfried

Moyrn, 2021-04-20:
Installed back 2.3.1
It works fine when disabling as you recommend.
i do not have in mind tweaking these for a long time. Possibly a thing I
keep pushing from legacy profiles.

At least I could slice and I will launch my print. many thanks.

Moyrn, 2021-04-20:
THanks Lukas

lukasmatena, 2021-04-21:
I'm glad that it helped, but I would keep the issue open. It should not hang.

closed 2026-09-29 by prusaslicer-bot[bot]