prusa3d/PrusaSlicer #6393: Change size nozzle

0scar8, 2021-04-20:
### Version
2.3.1+

### Operating system type + version
Windows 8

### 3D printer brand / version + firmware version (if known)
Nothing

### Behavior
In a new printer installation, when you change the size of the nozzle, the decimal digits are erased. Example: 0.25mm changes to 25.00mm

[image]

gurnarok, 2021-04-20:
I tried this, it doesn't take a comma as a valid input. You need to use dot.

0scar8, 2021-04-20:
I have used dot as in the photo.

rtyr, 2021-04-21:
Does it happen when you switch the language to English?

0scar8, 2021-04-21:
I have changed the language to English and it works fine. (see the photos)
In Spanish it has worked badly for me.

Thanks to all!!
Bye!

[image]
[image]

closed 2021-04-21 by 0scar8