prusa3d/PrusaSlicer #6395: Minimised window does not show up again on macOS 11.2.3 @ M1 Mac mini

drelich, 2021-04-20:
### Version
2.3.1+arm64

### Operating system type + version
macOS 11.2.3 on M1 Mac mini

### Behavior
A window minimised into dock does not show up again upon clicking the app's icon. The only way to display the window again is to right-click the icon, click 'Show All Windows' and then clicking on the window preview.

plucker, 2021-12-16:
I have this problem too.  The application often doesn't follow Apple UI guidelines in other areas.  For instance the PrusaSlicer menu should contain an "About PrusasSlicer" as the first item and also the application should not close when the red cross icon at the top left of the window bar is clicked.  It should close the window but the app should continue running like nearly all other OSX apps do.

J-charles-C, 2021-12-23:
I,

Same problem with 2.4.0 on OS X Big Sur.
Thanks for your job.
cordially

baensch, 2022-02-04:
Hi Together,
For me it is still a Thing in 2.4.0 arm64is there anything we can provide to the Devs to solve this issue?

plucker, 2022-02-04:
Yep, still happening here, it's annoying and it's not following the Apple standards, also the About PrusaSlicer and Preferences functions are not in the right menus and when you click o the close window icon at the top left it is closing the application rather than just the window.  These are all against Apple standard UI principles and should be changed.  It's actually quite difficult to write an application in the OSX environment without following these standards so the devs have built this in a slightly odd way.

baensch, 2022-02-04:
I use Prusa Slicer on 3 different Operating systems iam quite happy that they have the UI consistent (i dont checked all but it looks so for me in the first view). So from my site that is not the "Problem".

n8bot, 2022-02-04:
> Yep, still happening here, it's annoying and it's not following the Apple standards, also the About PrusaSlicer and Preferences functions are not in the right menus and when you click o the close [...]

Where should the about dialog and preferences be?

What should happen when the red button is pressed? Which button do you mean and what do you expect exactly?

Do you have a link to the standards about apple software?

plucker, 2022-02-04:
> Where should the about dialog and preferences be?

https://developer.apple.com/design/human-interface-guidelines/macos/menus/menu-bar-menus/
[image]

You've got most of this correct but are missing the about and preferences items off the App menu, they should be the first two items before services

> What should happen when the red button is pressed? Which button do you mean and what do you expect exactly?

[image]

Window and currently open file should close but App should remain running and showing in the dock with an "active" dot underneath it.  If you want to close an app the standard method would be PrussaSlicer > Quit PrusaSlicer or Command Q (which do work in PrusaSlicer but the red button shouldn't close the app, rationale being that you might want to immediately open another file in that app. https://www.howtogeek.com/279086/why-do-mac-apps-stay-open-when-i-hit-the-red-x-button/#:~:text=How%20to%20Quit%20Apps%20in,It's%20that%20simple.

> Do you have a link to the standards about apple software?

The first link gives you al the Human Interface guidelines

None of this is critical but is uncomfortable for OSX users to not be able to find the standard things where they normally are.

Jeff

baensch, 2022-03-13:
> ### Version
> 2.3.1+arm64
>
> ### Operating system type + version
> macOS 11.2.3 on M1 Mac mini
>
> ### Behavior
> A window minimised into dock does not show up again upon clicking the app's icon. The only way [...]

I Think you can Close this issue now :-) For me it is working with the [2.4.1](https://github.com/prusa3d/PrusaSlicer/releases/latest) Release. 
Thanks to the Devs :-)

J-charles-C, 2022-03-13:
I, 
With last Big Sur update and late 2014 intel  iMac, i have always the problem.

best

baensch, 2022-03-13:
> I, With last Big Sur update and late 2014 intel iMac, i have always the problem.
> 
> best

Do you installed already the 2.41 release of Prusa Slicer? I have a Macbook Pro 14" with a m1 Pro MacOS Monterey (12.2.1) and For me it is gone.

J-charles-C, 2022-03-14:
Yes.

closed 2026-09-29 by prusaslicer-bot[bot]