prusa3d/PrusaSlicer #6396: Plater 3D view not rendered (empty box), Unable to init glew library

fiechr, 2021-04-20:
### Version
2.3.1

### Operating system type + version
Manjaro Linux

```
PrusaSlicer
Version:   2.3.1
Build:     PrusaSlicer-2.3.1+UNKNOWN

Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 5.11.14-1-MANJARO x86_64
Total RAM size [MB]: 33,585MB
OpenGL installation
GL version:   4.6.0 NVIDIA 460.67
Vendor:       NVIDIA Corporation
Renderer:     GeForce GTX 1050 Ti/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
[... 417 lines cut ...]
[2021-04-20 19:08:06.530791] [0x00007f0f7790a080] [error]   Unable to init glew library
[2021-04-20 19:08:06.547425] [0x00007f0f7790a080] [error]   Unable to init glew library
[2021-04-20 19:08:06.548104] [0x00007f0f7790a080] [error]   Unable to init glew library
[2021-04-20 19:08:06.564653] [0x00007f0f7790a080] [error]   Unable to init glew library
[2021-04-20 19:08:06.565323] [0x00007f0f7790a080] [error]   Unable to init glew library
[2021-04-20 19:08:06.581868] [0x00007f0f7790a080] [error]   Unable to init glew library
[2021-04-20 19:08:06.582538] [0x00007f0f7790a080] [error]   Unable to init glew library
[2021-04-20 19:08:06.612745] [0x00007f0f7790a080] [error]   Unable to init glew library
[2021-04-20 19:08:06.613871] [0x00007f0f7790a080] [error]   Unable to init glew library
[2021-04-20 19:08:06.620009] [0x00007f0f7790a080] [error]   Unable to init glew library
[2021-04-20 19:08:06.620856] [0x00007f0f7790a080] [error]   Unable to init glew library
[2021-04-20 19:08:06.635473] [0x00007f0f7790a080] [error]   Unable to init glew library
[2021-04-20 19:08:06.636112] [0x00007f0f7790a080] [error]   Unable to init glew library
[2021-04-20 19:08:06.652494] [0x00007f0f7790a080] [error]   Unable to init glew library
[2021-04-20 19:08:06.653152] [0x00007f0f7790a080] [error]   Unable to init glew library
[2021-04-20 19:08:06.669538] [0x00007f0f7790a080] [error]   Unable to init glew library
[2021-04-20 19:08:06.670188] [0x00007f0f7790a080] [error]   Unable to init glew library
[2021-04-20 19:08:06.686552] [0x00007f0f7790a080] [error]   Unable to init glew library
[2021-04-20 19:08:06.687368] [0x00007f0f7790a080] [error]   Unable to init glew library
[2021-04-20 19:08:06.703737] [0x00007f0f7790a080] [error]   Unable to init glew library
[2021-04-20 19:08:06.704387] [0x00007f0f7790a080] [error]   Unable to init glew library
```

#### Project File (.3MF) where problem occurs
n/a (all)

bubnikv, 2021-04-21:
Does it happen with our builds?

Our general stance to Linux builds is the following:

We support only builds compiled by us or compiled using the static dependency scripts you find in the deps directory.

fiechr, 2021-04-21:
I understand.

The appimage version does work.

I guess I stay with 2.3.0 for now and wait for someone smarter than me to figure this out.

bnavigator, 2021-07-09:
Hi @fiechr,

on openSUSE Tumbleweed we ran into the same problem and fixed it by using the bundled GLEW 1.13.0 from PrusaSlicer instead of the system GLEW 2.2.0 library with wxWidgets 3.1.5. If you adjust the PKGBUILD and make sure that the GLEW headers are not installed during makepkg, this might work for Archlinux and Manajaro as well.

(*PrusaSlicer 2.3.1 and 2.3.2*)

bubnikv, 2021-09-25:
We only have limited resources to support Linux self-compilation using our static dependencies as described in 
https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md

closed 2021-09-25 by bubnikv

funkycode, 2022-12-30:
works on wx 3.2 with glew 2.2, with following patch:

```
diff --git a/src/slic3r/GUI/OpenGLManager.cpp b/src/slic3r/GUI/OpenGLManager.cpp
index a4d8d9a0f..3eb8333fb 100644
--- a/src/slic3r/GUI/OpenGLManager.cpp
+++ b/src/slic3r/GUI/OpenGLManager.cpp
@@ -234,7 +234,8 @@ OpenGLManager::~OpenGLManager()
 bool OpenGLManager::init_gl()
 {
     if (!m_gl_initialized) {
-        if (glewInit() != GLEW_OK) {
+        GLenum err = glewInit();
+        if (err != GLEW_OK && err != GLEW_ERROR_NO_GLX_DISPLAY) {
             BOOST_LOG_TRIVIAL(error) << "Unable to init glew library";
             return false;
         }
```