prusa3d/PrusaSlicer #6400: Feature request - closing holes in external perimeter if seam position is used

romanicak, 2021-04-21:
### Version
2.3.0

### Operating system type + version
Win10

### 3D printer brand / version + firmware version (if known)
MK3S

### Behavior
Feature request - closing holes in external perimeter if seam position is used.

When using the seam position connection when moving from the external perimeter to the fill (in this case the gap fill), the nozzle passes as if straight and pulls the filament path inwards. He will probably omit a piece of filament?
[image]
[image]

 Or I don't know, but it was such that a little hole would stay there. And here I do not want, I need to have a nice surface, if possible perfect. As if the head just didn't finish the perimeter where it was?

Retraction has no effect, various settings have been tested and no result.

Those holes can also be seen in the slicer, is that right?

During a discussion on another forum, it was found out:
In Cura, wipe while retrating works so that Cura moves the "seam" in the same direction, so it closes the hole.
But slic3r doesn't go through the hole in the same direction, but goes towards a new place (like an infill or a completely different place) and leaves a hole behind.

When I set the wipe in Cura it does move see. attachment. It will overshoot it, for example, by an extreme 5mm here. Which is great, because it just goes over and closes.
Imgur: The magic of the Internet
https://imgur.com/JIk33Ox

Nothing like this can be set in Slic3r. There it comes to see the photo and then immediately goes inside and goes to solve the infill ... So it sucks.
Imgur: The magic of the Internet
https://imgur.com/3gaMagR
This can be partially bypassed by setting the "infill before perimeters", so in most cases the slic3r does not go in after the outer wall ends, but goes out somewhere, but you don't want to, because infill is then visible in outer walls....

#### Project File (.3MF) where problem occurs
ValecTest_DxH_30x15.3mf zipped
[image]
[image]
[ValecTest_DxH_30x15.zip](https://github.com/prusa3d/PrusaSlicer/files/6348865/ValecTest_DxH_30x15.zip)

murk-sy, 2021-04-21:
I am having this issue too, and it's frankly a bit frustrating - I am forced to use seam painting on completely invisible surfaces as there ends up a lot of material missing due to what I imagine is wipe related issues (using pretty much exclusively PETG - to be clear I have K factor calibrated). Or, alternatively, I avoid using seam painting at all and instead use seams aligned back and rotate models.

I have verified that the amount extruded on seam painted and non seam painted parts is exactly the same. The issue is likely the wipe, which does not happen the same way when internal and external seams are not aligned.

See also issues #5129 (the first few part images show basically the exact issue I'm having, except it's worse for me), #6186, #5149 - aligning internal and external seams would likely resolve all these issues and improve bumps where internal seams are used.

romanicak, 2021-04-21:
> I am having this issue too, and it's frankly a bit frustrating - I am forced to use seam painting on completely invisible surfaces as there ends up a lot of material missing due to what I imagine is [...]

I see, maybe the same problem. Head move is not finising the actual line or trace, but moves directly to GapFill/InternalPerimeter/Infill. To implement solution like Cura has should solve this issue. We all know that Prusa Printers deserve to be perfect and print quality prints, as they are presenting it, right? :-)

romanicak, 2021-04-26:
Test model is only cylinder, what i need to print seamlessly is here.
[Grdw_Plovak.zip](https://github.com/prusa3d/PrusaSlicer/files/6377461/Grdw_Plovak.zip)

murk-sy, 2021-04-29:
New issue report with same problems #6448

Though this issue isn't the first created on this, I'll try to at least link duplicates here so it's sort of centralized.

I wish back-left-edge was also an option in seam positions since that would make my life a little bit easier right now.

Launchpad-McQ, 2021-08-28:
I am experiencing this problem too. Is there a clear way forward on how "wipe for external perimeters" can be implemented in PS?

murk-sy, 2022-08-10:
The gap issue on seam painting has been resolved for me by using Arachne slicing engine included with 2.5.0 alphas

DuckersMcQuack, 2022-10-17:
> The gap issue on seam painting has been resolved for me by using Arachne slicing engine included with 2.5.0 alphas

I use that as well, but compared to cura's "wall wipe distance", it's still so ugly and uneven. and prusa has practically no settings other than "where do you want to place it" No advanced settings to tune wall wipe, nor even coasting to compensate for no extra walls to wipe on. Just jumped over from cura as cura is severely misbehaving.

[image]
[image]

closed 2026-09-29 by prusaslicer-bot[bot]